Denvi/Candle #75: grblControl going all haywire

classbproject, 2016-10-05:
Starting yesterday my installation of grblControl has been acting all crazy. First of, the application is very very slow. Opening windows for file selection takes anywhere between 5 to 10 seconds. Also the visualization is not working anymore. Also noticed a few grbl decode errors as well. For e.g. I had set G90 as the active distance mode but when I started the machine it followed G91. Can you please help me debug?

Denvi, 2016-10-05:
Hello.
With CNC disconnected/turned off grblControl also works slow, visualizer doesn't works?
Post "settings.ini" file.

classbproject, 2016-10-05:
In the visualizer the pendant remains stationary at 0,0. Attached here is the settings.ini. Also circles are being drawn as ellipses. Also I noticed that another program (not CNC related) that was recently installed, also installed Microsoft Visual C++ 2013 redistributable 12.0.30501. with it. I removed this and noticed that the problem was fixed. However it is now happening again.
[settings.ini.txt](https://github.com/Denvi/Candle/files/511368/settings.ini.txt)

Denvi, 2016-10-05:
> With CNC disconnected/turned off grblControl also works slow, visualizer doesn't works? 

Still wait answer?

Settings looks ok.

classbproject, 2016-10-05:
Yes with the CNC disconnected it still works slow.

Denvi, 2016-10-05:
What PC specs (CPU, GPU)?
Whether change of FPS setting to 30 changes behavior?

classbproject, 2016-10-05:
PC is a Pentium dual core 2.1 GHz, 6GB ram, GPU is an Intel integrated one. I didn't change the fps. Let me change that and see. I'll keep you posted.

classbproject, 2016-10-07:
A few more updates. When the application finds the previously configured serial port active then its not slow however, it still does not 'pause' and 'abort'. Also the pendant just keeps rotating at 0,0 but does not animate.

Denvi, 2016-10-07:
$$ - command output please.

classbproject, 2016-10-07:
Here you go:

```
Grbl 0.9j ['$' for help]
>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=80.000 (x, step/mm)
$101=80.000 (y, step/mm)
$102=80.000 (z, step/mm)
$110=6000.000 (x max rate, mm/min)
$111=6000.000 (y max rate, mm/min)
$112=6000.000 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=127.000 (z max travel, mm)
ok

```

classbproject, 2016-10-08:
I think I have resolved this. There was a 32-bit copy of `Microsoft Visual C++ 2013 redistributable 12.0.30501` installed on my machine along with the 64-bit one too. I removed them both and now everything seems to be working fine. I'm closing this issue for now.

closed 2016-10-08 by classbproject