Denvi/Candle #77: How to get candle or grblcontrol installed in raspberrypi

jvarl, 2016-10-08:
Hi, in previous post (see #55) I was trying grblcontrol over different pc's. Also I tried to install this friendly program on my raspberry pi without success (see #72). This time I announce that finally my efforts were crowned with success. I have installed grblcontrol and Candle in my raspi-3b raspbian on two different images, one in the traditional Jessie (2016-05-27-raspbian-jessie.zip image) and another in the new jessie PIXEL. In both cases I used micros-sd memory of 16GB Class 10. I want to provide information about steps I walked for people interested in getting it and do not know how to achieve it. I made an installation of libraries QT5 by native compilation of sources from the git repository cloned on raspbian (to compile source code Denvi, qt5.4 or higher is required). I rely on recommendations found in the QT wiki post “Native Build of Qt5 on a Raspberry Pi”. In my case, obtain QT5 libraries compiled from source required a time of 8 hours in the raspi-3b (mentioned article refers 2 days?). While I followed meticulously and carefully the steps there, trying not to make mistakes, I needed to modify the data about type and amount of binary packages that I installed, because from 56 packages given there only 15 I used to make my 203 proper packages installed, based on tests I did. Do not use the package list in the post nor did the patch give there. Only use “my-package-list”, “sudo apt-get install <my-package-list>”, then “sudo apt-get update”, “sudo apt-get upgrade”. Following I give “my-package-list” to install previous to clonate git qt5 sources, init-repository… , configure…, make…, sudo make install… etc.etc. Pay attention for changes to introduce in files there mentioned. Use “copy-paste” for text on the post and edit every line in your preferred text editor previously to paste it in the terminal command line to be executed. Pay attention using “sudo” were needed and change use permissions for private files. If you intend compile Candle in qt-creator you must fix two present errors in the source code gave at present as .zip file. In frmmain.h you must append the following line: #include <QGLFormat>, and in qlwidget.cpp you must append: #define GL_PROGRAM_POINT_SIZE 100.
“my-package-list”:
autopoint comerr-dev debhelper docbook-xml firebird-dev firebird2.5-common firebird2.5-common-doc firebird2.5-server-common freetds-common freetds-dev gettext gir1.2-gst-plugins-base-0.10 gir1.2-gstreamer-0.10 gparted gstreamer-tools gstreamer0.10-gconf:armhf gstreamer0.10-plugins-bad:armhf gstreamer0.10-plugins-good:armhf gstreamer0.10-tools gstreamer0.10-x:armhf icu-devtools intltool-debian krb5-multidev libasound2-dev:armhf libasprintf-dev:armhf libatkmm-1.6-1:armhf libavahi-client-dev libavahi-common-dev libavcodec-dev libavformat-dev libavresample-dev libavutil-dev libbotan-1.10-0 libcairo-perl libcairomm-1.0-1 libcdaudio1 libct4:armhf libcups2-dev:armhf libdbus-1-dev:armhf libdirac-encoder0:armhf libdrm-dev:armhf libdrm-exynos1:armhf libdrm-omap1:armhf libdrm-tegra0:armhf libept1.4.12:armhf libfbclient2:armhf libfbembed2.5 libfontconfig1-dev:armhf libgettextpo-dev:armhf libgettextpo0:armhf libglib-perl libglib2.0-dev libglibmm-2.4-1c2a:armhf libgssrpc4:armhf libgst-dev libgst7 libgstreamer-plugins-bad0.10-0:armhf libgstreamer-plugins-base0.10-dev libgstreamer0.10-dev libgtk2-perl libgtkmm-2.4-1c2a:armhf libib-util:armhf libicu-dev:armhf libiodbc2:armhf libiodbc2-dev libjpeg9:armhf libjpeg9-dev:armhf libkadm5clnt-mit9:armhf libkadm5srv-mit9:armhf libkdb5-7:armhf libmail-sendmail-perl libmpcdec6:armhf libmysqlclient-dev libmysqlclient18:armhf libpango-perl libpangomm-1.4-1:armhf libparted-fs-resize0:armhf libpcre3-dev:armhf libpcrecpp0:armhf libpq-dev libpq5:armhf libpthread-stubs0-dev:armhf libpulse-dev:armhf libpulse-mainloop-glib0:armhf libqt5clucene5:armhf libqt5concurrent5:armhf libqt5core5a:armhf libqt5dbus5:armhf libqt5declarative5:armhf libqt5designer5:armhf libqt5designercomponents5:armhf libqt5gui5:armhf libqt5help5:armhf libqt5network5:armhf libqt5opengl5:armhf libqt5printsupport5:armhf libqt5qml5:armhf libqt5quick5:armhf libqt5quicktest5:armhf libqt5quickwidgets5:armhf libqt5script5:armhf libqt5sql5:armhf libqt5sql5-sqlite:armhf libqt5test5:armhf libqt5webkit5:armhf libqt5widgets5:armhf libqt5x11extras5:armhf libqt5xml5:armhf libqt5xmlpatterns5:armhf librarian0 libsigsegv2:armhf libslv2-9 libsqlite0 libsqlite0-dev libsqlite3-dev:armhf libssl-dev:armhf libssl-doc libswscale-dev libsybdb5:armhf libsys-hostname-long-perl libudev-dev:armhf libunistring0:armhf libvte-2.90-9 libvte-2.90-common libx11-dev:armhf libx11-doc libx11-xcb-dev:armhf libxau-dev:armhf libxcb-glx0-dev:armhf libxcb-icccm4:armhf libxcb-icccm4-dev:armhf libxcb-image0:armhf libxcb-image0-dev:armhf libxcb-keysyms1:armhf libxcb-keysyms1-dev:armhf libxcb-randr0:armhf libxcb-randr0-dev:armhf libxcb-render-util0:armhf libxcb-render-util0-dev:armhf libxcb-render0-dev:armhf libxcb-shape0-dev:armhf libxcb-shm0-dev:armhf libxcb-sync-dev:armhf libxcb-sync1-dbg:armhf libxcb-xfixes0-dev:armhf libxcb-xinerama0:armhf libxcb-xinerama0-dev:armhf libxcb-xkb1:armhf libxcb1-dev:armhf libxdmcp-dev:armhf libxext-dev:armhf libxfixes-dev:armhf libxi-dev libxkbcommon-x11-0:armhf libxml2-dev:armhf libxrender-dev:armhf libxslt1-dev:armhf libzvbi-common libzvbi0:armhf mysql-common po-debconf qml-module-qtquick-controls:armhf qml-module-qtquick-layouts:armhf qml-module-qtquick-window2:armhf qml-module-qtquick2:armhf qt5-doc qtbase5-dev-tools qtbase5-doc qtcreator qtcreator-data qtcreator-doc qtdeclarative5-dev-tools qtdeclarative5-doc qtenginio5-doc qtgraphicaleffects5-doc qtmultimedia5-doc qtpositioning5-doc qtquick1-5-dev-tools qtquickcontrols5-doc qtscript5-doc qtsensors5-doc qtserialport5-doc qtsvg5-doc qttools5-dev-tools qttools5-doc qttranslations5-l10n qtwebkit5-doc qtwebkit5-examples-doc qtwebsockets5-doc qtx11extras5-doc qtxmlpatterns5-dev-tools qtxmlpatterns5-doc rarian-compat sgml-data synaptic x11proto-core-dev x11proto-fixes-dev x11proto-input-dev x11proto-kb-dev x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
If it were of interest to get a raspbian image with qt5.7.1 libraries installed and grblcontrol and Candle, please give me to know and I will introduce a link to download (dropbox or google drive). Best regards. JVARL

Denvi, 2016-10-08:
Hi, how it works on Raspberry?
Some time ago I compiled the program for the 2 version of a board.
But the program worked not in the best way. There were problems with visualizer/sender performance.

jvarl, 2016-10-08:
With my CNC hardware setup it is working fine, like as runs bCNC and UGCS that I tried in raspian jessie too. Because it uses opengl it deploys only at full size window.

pihnat, 2016-10-14:
Hi jvarl
I'd love to try Candle on a raspi. Could you supply a link to download an image or maybe setup apt-get so I can install to a Jessie setup (with Pixel)?

jvarl, 2016-10-14:
Hi Pihnat, As soon achieve upload the image to the cloud (google drive) I will include the link. This may take some time, since the compressed file (7z) weighs about 5GB. The first thing you should do when you have the image in your raspi-3b is set raspbian with data of your local area, language, keyboard and wifi. Mine is America, Argentina, Spanish, Buenos Aires

jvarl, 2016-10-18:
Hi pihnat, I apologize for delay in answering. I lost a lot of time by faults in provision of internet and energy supply. After that I found several difficulties associated with uploading to the cloud a very large .7z file size, I decided to upload it in the form of smaller pieces. I did so, in order to achieve better control of progress uploading files. You will need to download and rebuild the large .7z file on the basis of small files, for which, I included details of how to do it as a README paragraph onto the .txt file included. The link for download is [https://drive.google.com/open?id=0B6LOs5zWeC-gLXR3QW1EUmt3Qzg] . While I remain for comments, I hope you enjoy the work I made. Best regards. JVARL.

pihnat, 2016-10-18:
Hi jvarl
I thought the large file size might have caused a problem with uploading but looks like you got around it. It will take me a little while to download all the files but I will let you know how it goes.
Thanks for your efforts, I'm looking forward to trying it out.

pihnat, 2016-10-19:
Wow, I must say congratulations! It works very well. Here's some feedback:

I'm using a raspi 3.
1. the 3D manipulation is great, very smooth when you zoom in/out or rotate. This is where I thought there might be a speed problem with the raspi but it works surprisingly well.
2. in 'CHECK' mode the mouse becomes sluggish. But that's OK because you wouldn't normally do much with the 3D view while it's doing the check. I tried it with a G-code file containing many lines and a few arcs as well as one which had many arcs (40,000 lines). The mouse is very sluggish in the latter case.
3. during a normal Send ie while cutting, the mouse and 3D manipulation is good. 
4. there's a few quirks I found:
a:  after you run a CHECK or after cutting if you go back to the Service Menu it won't let you change settings like Query period, Rapid speed, etc. You have to quit Candle and re-start it.
b:  when you quit Candle there are a few funny things happening. On one occasion when I quit the web browser was open and connected to the qt site. On other occasions 2 FileManager windows were open, one was /home/pi and the other was /home/pi/.themes.

Overall it's great to see that Candle can run on the raspi and congratulations again for your efforts. I will continue with feedback and I hope others try running it on the raspi too.

jvarl, 2016-10-20:
Hi pihnat, I am happy that you also enjoy this fantastic “like nix” operating system which is raspbian-jessie-pixel, also the powerful Qt5 programming libraries and IDE, at join with this friendly grbl controller what is “Candle” by Denvi. But, in my land is often said "Not everything that glitters is gold", and you must take these “as it is, at your proper risk”. So, I would like give some advice to you and people that could read the post. Sometime, I experienced that full screen got all blank color, without possibility of control the things. In this case, you must just press “ctrl-alt-backspace” and screen newly appears (I remembered older situations of “blue screens” in a popular OS). First, change your password and if possible user name too for enhance security against outer attacks. Second, using qtcreator IDE compile a fresh copy of Candle as a new project with a folder name that you only known. Third, if possible try to use Candle without 3w connection. If you were to experience case such I have mentioned, you will do “clean” and “recompile” the candle project in two situations, first on “debug” then same in “release” (on debug ever is responsive positively, because there not build executable, just work internally). For me, I have got success doing that, trying the project over a new folder. Finally, using mouse-right-button on candle-desktop-launch-icon (properties->entry_to_desktop->new_path_to_binary_executable) you can update the simlink to the new folder. Do not forget to introduce the two changes at the current source code, as I mentioned in previous entries. Sincerely: JVARL.

pihnat, 2016-10-21:
Hi jvarl,
Thanks for the extra information. I hope to re-compile Candle after a few modifications. I have done this under Windows so it will be interesting to try it under Raspian. A few comments:
1. to run in Windows grblControl/Candle has all files in a folder which can be placed anywhere and a shortcut created to the exe file, no installation required. I have basic linux knowledge but can the same be done with Candle under Jessie, ie can I just copy the Candle release folder to another raspi and run it there without installing it or having Qt installed?
2. when you uploaded the img file of Jessie/Qt/Candle would it have been OK to use Jessie lite instead of Jessie with Pixel? I'm thinking that the size of the img file would have been much smaller because there would be no LibreOffice, etc.

BTW I'd like to share with you a photo of the program running with the raspi 7" touch screen. Since it's only 800x480 there's some issues with seeing the whole control panel properly, eg in Settings you can't see or get to the OK button. But that is all fixable by re-arranging buttons, fonts, etc and re-compiling.

Many thanks again, you have started a new method of using grblControl/Candle which I'm sure will be expanded on.
regards
Peter

[image]

sico33, 2016-10-21:
Hola jvarl, disculpa mi ignorancia pero vengo siguiendo el hilo del tema y me puse muy contento al saber de tu trabajo.
Mi proyecto sobre la construcción de mi propio CNC sigue cada vez mas cerca.
No tengo conocimientos en Rasp y su sistema operativo, pero cada día aprendo mas.... toda la estructura la vengo construyendo saliendo desde el punto CERO ----ni siquiera sabia que existía Arduino ----

Necesito pedirte me puedas ayudar para poder ejecutar el Soft y así probarlo con la placa---

Aun estoy sin éxito de que la aplicación se ejecute --- ya probé con una instalación limpia nuevamente y todavía no logro ejecutarla.

Me podrías dar una mano???

Desde ya muchas gracias y saludos.

PD: Soy del interior de la prov. de Buenos Aires

[image]

pihnat, 2016-10-21:
Hi sicco33,
if you double click the Candle shortcut on the desktop the program should run. Which raspi are you using and are you using a normal monitor plugged into the hdmi port on the pi?

sico33, 2016-10-21:
Estoy usando un RSpi 3 con VNC.....intentare realizar la prueba con HDMI.....ya ejecute el doble clic sobre el acceso directo pero no se ve la interfaz, si veo que el proceso se ejecuta por eso adjunte la imagen.

Voy a realizar la prueba con el monitor y realizo mis comentarios.

Saludos !!!

pihnat, 2016-10-21:
A HDMI monitor (or VGA with a HDMI to VGA converter) is the way to run it on a raspi. Plug a USB mouse and keyboard into the raspi too.

But NOTE .... the raspi version is experimental and has some bugs. If you need a useable solution you should be using grblControl/Candle on a PC. It is much more reliable especially when connected to a CNC machine - you don't want the program crashing in the middle of machining or the CNC machine losing control. The PC version of grblControl has been well tested and de-bugged.

Since you are starting out and learning about CNC control + building a machine could I suggest that you use the PC version first until you become more experienced. It's a lot to get your head around Arduinos, grbl, grblControl, G-Code, etc and knowing that the control program on a PC works well is one less headache.

sico33, 2016-10-21:
pihnat muchas gracias por tus consejos....pude probar el sistema conectandolo a un monitor HDMI y todo corre OK.
Voy a seguir investigando y probando con las difentes configuraciones del Soft y GRBL.

Muchas Gracias por todo !!!

Saludos!!!

jvarl, 2016-10-21:
Hi, pihnat
1. I do not think this possible. The executable binary file is not self content to run itself (not compiled in a completely static way), there are shared libraries (dynamic), called at runtime within the complex structure of QT5 (and the operating system itself). Try this could be simple and entertaining. I did not try this yet.
2. You can choose from a minimal distribution and then add packages or a complete distribution and uninstall what is not used (it is simpler). On the Internet, the latest is advised for economy of effort. You have to think that installing libraries QT5 alone weighs about 3.5GB.
Regards. JVARL

jvarl, 2016-10-21:
Hola sico33
Un gusto en saludarte y decirte que en la medida de que pueda estoy dispuesto a ayudarte. Sería deseable que nos tuvieras al tanto de tus avances y dijeras como está constituido tu CNC hardware setup. Me alegro que pudieras hacer funcionar la imagen raspbian-jessie-pixel-candle en tu raspi. Saludos. JVARL

jvarl, 2016-10-24:
Hi, pihnat.
In this last days I tried installing qt5 libraries starting from lite in a 8GB micro-sd, but for giving to myself a good living I installed the light weight LXDE GUI (lite has not any GUI at all), and using iceweasel (firefox) instead of chromium. The initial starting size of lite in such condition was as only of 1.9GB. After installing my-package-list (base support for git clone qt5 and get it to run) the size was 2.6GB. Then I ran git cloning QT5 and “./init-repository”, which became the size 6.2GB (only 1GB less than the total size of the micro-sd, which was 7.2GB). This forces me to "change horses in midstream" and had to move to a micro-sd 16GB to continue the test. After of “./configure”, and previous to do “make”, the occupied size was 6.5GB. Again, after I did ”make &> output_make”, the occupied size became 10.1GB (free space 4GB from a total 14.1GB space of my 16GB micro-sd). If I compare the free space of 4GB obtained against 1.8GB in the previous case of using the complete raspbian-pixel, I arrive to the thinking that very probably I will get a similar free size (do not tried yet) starting from the latest, and removing everything what I do not need at all. Additionally in this case I remain using a more comfortable raspbian distribution. BTW. Firefox and chromium are similar. Keeping using chromium is a good choice. Regards. JVARL.

pihnat, 2016-10-27:
Hi jvarl,
I did the following but am now stuck:

SD card size 14.6G.
Installed Jessie Lite then
sudo apt-get update && sudo apt-get upgrade
sudo apt-get install xinit
sudo apt-get install lxde-core lxterminal lxappearance
sudo apt-get install lightdm                                  now 12.5G on SD card free
sudo apt-get install chromium-browser                 12.3G free
sudo apt-get install qt5-default                             12.2G free
sudo apt-get install qtcreator                                11.9G free
Using Chrome I downloaded Candle and unzipped it. Qt runs great and Candle source code opens OK. I also appended the lines you mentioned in frmmain.h and qlwidget.cpp

I had no problems under Windows but there are 2 problems on the raspi:
- how do you install MinGW compiler and
- what about the serial port driver?

regards

jvarl, 2016-10-27:
Hi pihnat, you can’t install qt5-default (see #72), hence arises the need to install qt5 cloning from github. The first steps you did are fine but only you missed install xserver-xorg too. Your option by chromium is OK. I haven’t clear why you mention mingw, which in linux isn’t needed at all. Within “my-package-list” that I gave, are included qtcreator, qtcreator-data, qtcreator-doc and qtserialport5-doc packages, so you don’t need install them separately, just “sudo apt-get install <my-package-list>”. Also, as “lite” is an absolutely minimal distribution, you should install git (“sudo apt-get install git”) before to “git clone anything”. You can optionally omit installing gparted and synaptic for saving an extra space. Even more, you don’t need a 3w browser at all for proceed, but it could be commode for you to read from an open window steps to follow when installing qt5 and you can copy candle.zip on raspi using pendrive. Regars. JVARL

pihnat, 2016-10-28:
Hi jvarl, thanks for your reply. I'm not a Linux person so am a little lost.
I need mingw because in my installation of Qt5 there is no compiler specified. Also Denvi mentions under "Build requirements" on the main page of the repository that it requires "Qt5.7.0 with MinGW compiler".
BTW I had to install mingw separately with the  Windows version of Qt5 I've been using too.

h2mille, 2016-10-28:
Hello,
thanks for this very nice project. I'm trying to make it work well too on my RPi, and I have the problem you solved, pihnat: I'm trying to change the resolution of the fullscreen without success. I never used qt before, but reach to recompile the project, changing resolution in ui, but no difference. can you tell me the magic?

jvarl, 2016-10-28:
Hello h2mille, Tell me what you currently have installed on raspi. From what I understand, the image raspian with qt5.7.1 libraries and you cannot operate your touch screen 7" (800x480) after changing the graphics resolution, that is rigth? You could explain further what you have added/modified in source code candle project (and where). I'm like you in using qt (I have only basic knowledge), but I programmed GUI applications with GTK3+ libraries. I also would learn whether pihnat respond to you about the matter. But for this you have to specify clearly about steps you have made. Regards. JVARL

h2mille, 2016-10-28:
Hello jvarl,
I used the image [https://drive.google.com/open?id=0B6LOs5zWeC-gLXR3QW1EUmt3Qzg] you provided. I have an old hdmi screen with native resolution of 1280x800 that I want to use with my router, with the raspi. The image works directly well, I just update the resolution in /boot/config.txt).
What I did is mostly nothing, after checking the source code and discovering the qtcreator, I tryed to change: frmmain.ui setting line 9 and 10 into :     <width>1280</width> <height>800</height>, and compile it again (I need to select the compiler, I set gcc) but it's quite naive.
I also tryed to understand what's the matter of openGL with Qt and windowed mode, but I have no clue, and failed finding something useful.
Thanks for your help!

pihnat, 2016-10-28:
Thanks jvarl, I tried to remember how I installed things and forgot to list the xserver line. I have edited the post.

Hi h2mille,
Same problem, I have limited knowledge of Linux and Qt. I have been trying a few different things but I should summarise what I'm playing with:
1. the reason I tried Jessie lite is only so after Candle is working the img of the SD card is small so it's easier to share, that's the only reason
2. it would be good to change the size of buttons, etc so Candle's GUI would fit on the raspi LCD screen, resolution is 800x480
3. this is more important for me since I will be running Candle on a pi with a larger LCD monitor - to be able to put or install Candle on other raspi's without having Qt5 installed, ideally like Windows where you just put the Candle folder anywhere and run the exe file.

davezdesignz, 2017-04-01:
I recently installed this [https://drive.google.com/open?id=0B6LOs5zWeC-gLXR3QW1EUmt3Qzg] on a raspberry pi 2 model b+ and the program opens but when I connect my grbl CNC (uses an arduino mega - grbl 1.1) I get some strange characters in the Z axis machine coordinates after the 0.000 and the status comes up with unknown. I can use candle 1.1.5 on my windows laptop to run the CNC with no issues (that I have found yet). Any ideas what the issue could be? Has anyone been able to make the screen work for a 7" display?

thank you

downsun, 2017-04-04:
我下载安装了jvarl 提供的镜像，发现一个问题，不论我在通过什么客户端启动程序：Xwindow clent or mstsc or vnc.
grbl 都运行在显示器的窗口，应该是编译的时候设置的问题，其他程序就没有这个问题。

pihnat, 2017-04-04:
Hi davezdesignz,
Unfortunately I haven't had time to continue experimenting with grblControl on the raspi. But if you check my post on Oct 2016 you'll see grblControl running on the raspi 7" touchscreen monitor. It wasn't stable however and menus are squashed so the UI would need to be modified for it to work well.

Hi downsun,
If I translated your message correctly you are trying to run grblControl on the raspi using a remote client as the interface. This won't work. You need to plug a HDMI monitor and mouse/keyboard into the raspi and use it that way.

downsun, 2017-04-20:
Thank you very much.

PNixx, 2018-04-04:
I try compile Candle for raspberry pi and I see error: 
```
$ qmake
Project WARNING: GL ES detected. VAO will be disabled.
```
How I can fix it?

jvarl, 2018-04-17:
Hi PNixx. This message is simply a warnning not an error. You'll have to get around compiling the Candle source using qt-5.7.0 (GUI) what is present into the downloable raspbian image I provided , as it is previously instaled. Regards, JVARL.

AlexeyKichev, 2019-02-19:
Я не могу собрать Candle na Raspberry Pi 3 из за GL. В новой версии присутствует GLES 3!

1f7, 2019-02-19:
@AlexeyKichev 
`sudo apt install libgles1-mesa-dev`

jvarl, 2019-03-03:
At the current time the thread is closed

closed 2019-03-03 by jvarl

WZ9V, 2020-02-21:
I have not fully tested but I was able to get it to compile and run on a Raspberry Pi 4 4GB by installing Qt onto Raspbian 2/23/2020
sudo apt-get install qtcreator qt5-default libqt5serialport5-dev

Make the code changes from above

    In frmmain.h you must append the following line:

#include <QGLFormat>

    and in widgets/qlwidget.cpp you must append:

#define GL_PROGRAM_POINT_SIZE 100

qmake Candle.pro
make

./Candle to run.  

Hope this helps

pihnat, 2020-02-22:
Thanks for the update WZ9V. I'm still not that proficient with linux but when I run qmake I get the same message as PNixx on April 5, 2018, ie

$ qmake
Project WARNING: GL ES detected. VAO will be disabled.

How do I get past that warning?

pihnat, 2020-02-23:
OK problem solved, the message is definitely just a warning. I actually did have an error too where I typed
#define GL-PROGRAM_POINT_SIZE 100   instead of
#define GL_PROGRAM_POINT_SIZE 100

I found it easier to run QtCreator & loaded 'candle.pro'. I edited those 2 files there and then did the build. So Candle runs on my raspi3B+ but unfortunately I don't have any arduinos here at home (I have 2 at work) so have to wait till tomorrow to finish testing.
Thanks again.

WZ9V, 2020-02-24:
> Thanks for the update WZ9V. I'm still not that proficient with linux but when I run qmake I get the same message as PNixx on April 5, 2018, ie
> 
> $ qmake
> Project WARNING: GL ES detected. VAO will be disabled.
> 
> How do I get past that warning?

You can safely ignore that warning.

pihnat, 2020-02-26:
I have Candle running on a raspi3B+ now. First thing you need to do is change the colours used in the visualisation because the default is black for everything.
The display runs at about 10fps but when sending a G-code program to the CNC machine it drops to 8fps. Overall operation is a bit sluggish but the fact that it runs at all is great.
I'll try it on a raspi4 and will report back.

pihnat, 2020-03-07:
Just updating - I have Candle running in a 2G raspi4. It's actually works extremely well, not sluggish at all, in fact it runs as fast as on a PC. The frame rate varies between roughly 50 and 70fps. As I mentioned in my last post you need to change the colours in the visualisation so they're not all black. I also changed the Status Query Period to 40ms so the tool movement is smoother.

[image]

On a PC you would place the folder with all the Candle files somewhere like
C:/Program Files (x86)
and then put a shortcut to Candle.exe on the Desktop. No installation is required.
On the Raspi4 I've been able to do a similar thing - after building from the source code using qt5 I've copied the folder with the Candle files onto a different raspian SD card which doesn't have qt installed and it runs without problems.
If anyone's interested I can post instructions on how to get it working.

Corteil, 2020-03-30:
H pihnat, instructions would be great, I'm a little lost at the moment :-) or a download link to your working Candle folder.

pihnat, 2020-03-30:
Hi Brian,
Yes, I will put something together soon.

xingqianlilu, 2020-04-10:
Hi pihnat,may I ask a question.First,thanks for your sharing.Candle have been running on my pi3b+,I feel very happy.But,it shows unconect now.I do not now why?How can I setup the file setting.ini.When I download the candle project for pi4 from https://github.com/pihnat/Candle,I find that there is not a setting.ini file in the project.How can I candle this problem.Thanks very much.

xingqianlilu, 2020-04-10:
@jvarl @pihnat 
Hi jvarl,Hi pihnat,may I ask a question.First,thanks for your sharing.Candle have been running on my pi3b+,I feel very happy.But,it shows unconect now.I do not now why?How can I setup the file setting.ini.When I download the candle project for pi4 from https://github.com/pihnat/Candle,I find that there is not a setting.ini file in the project.How can I candle this problem.Thanks very much.

pihnat, 2020-04-10:
Make sure the Arduino is plugged into your raspi.
In the main menu click on **Service** then select **Settings**. Select **ttyACM0** as the port and set Baud rate to **115200**. When you click **OK** at the bottom of the page it will create a setting.ini file and save the settings there.

[image]

Note that Candle runs sluggishly on a pi3, I definitely recommend running it on a raspi4.

pihnat, 2020-04-10:
I just checked and settings.ini is in the zip file. Note that if you download the project to build using qt there will be no settings.ini until you run Candle the first time.
However if you download & unzip the Candle_1.1.8.zip file which is _ready to run_ on a raspi then there will be a default settings.ini file already in the Candle folder.

xingqianlilu, 2020-04-10:
OK，it is great , thank you very  much.Now,as you say ,I lanuch Candle and move right!However,it runs sluggishly as you say.I open a 40000 gcode file which spend 1min to open.Wheth it runs very smootly on pi4?

xingqianlilu, 2020-04-10:
[image]

pihnat, 2020-04-10:
1. The screenshot you posted is how Candle looks if there is no settings.ini file. Under **Servic**e, **Settings** you would need to select colours for background (eg white), Tool, Start Point, etc. When you press OK those colours will be saved. That is why I supplied a default .ini file with colours already selected.

2. On raspi4 I loaded a 36,400 line G-code file and it loads in less than 2 secs.

xingqianlilu, 2020-04-10:
Thanks  for  your  patient   reply.My pi3b+ is  2fps  and  it is  so sluggishly. My test  file  is 2M which is loaded from SD card.The time spent loading file is about  2 min.I think I need buy a pi4 and test the speed.The picture below is my candle parameter when run on pi3b+.2fps.
[image]

pihnat, 2020-05-13:
Note - I changed the github site from
**pihnat/Candle** to **pihnat/rpi-Candle**

ddllpp, 2024-09-08:
Has anyone had success installing this on a Raspberry Pi 5?
I've tried the pre-built installs and also building from source following the wiki pictorials.
The release does run using './Candle2', but the visualizer panel is empty and you see the desktop behind the window through it.
I've used the mesa override to get other software to run on the pi5 so I tried:
'sudo MESA_GL_VERSION_OVERRIDE=3.3 ./Candle2'
It runs, and now the visualizer is black, but you don't see any USC icon or anything, unlike the post above. Nothing but black when loading a project or changing the colours in settings.
Any thoughts?
Thanks very much.

mar0x, 2024-09-08:
@ddllpp Have a look at the forked [v1.1.9 release](https://github.com/mar0x/Candle/releases/tag/v1.1.9) with 64-bit [Candle-1.1.9-RPi-aarch64.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-RPi-aarch64.AppImage) and 32-bit [Candle-1.1.9-RPi-armhf.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-RPi-armhf.AppImage).

uzfm, 2025-01-19:
Інструкція з встановлення та запуску Candle - програми для керування GRBL з візуалізацією G-коду

Встановлення необхідних пакетів:
Оновлення списку пакетів:
Bash

sudo apt update
Встановлення Qt Creator та необхідних бібліотек:

sudo apt install qtcreator qtbase5-dev qt5-qmake g++ libqt5serialport5 libqt5serialport5-dev
Клонування репозиторію Candle:
Переміщення до бажаної директорії:

cd /home/uzfm/Downloads
Клонування репозиторію:

git clone https://github.com/Denvi/Candle.git
Збірка та запуск Candle:
Перехід до директорії проекту:

cd rpi-Candle-master/src
Створення Makefile:

sudo qmake candle.pro
Збірка програми:

make

Запуск програми:
./Candle

Вирішення проблеми з чорним екраном при візуалізації G-коду:
Причина: Можлива проблема з графічним бекендом.
Вирішення: Встановлення бібліотек OpenGL ES для забезпечення апаратного прискорення графіки.
Команда для встановлення:

sudo apt install libgles2-mesa-dev
Запуск програми з використанням програмного рендерингу (якщо апаратне прискорення не працює):

LIBGL_ALWAYS_SOFTWARE=1 ./Candle
Додаткові пояснення:
OpenGL ES: Це стандарт для 3D графіки, який часто використовується в мобільних пристроях та вбудованих системах. Candle використовує OpenGL ES для візуалізації G-коду.
LIBGL_ALWAYS_SOFTWARE=1: Цей параметр змушує програму використовувати програмний рендеринг замість апаратного. Це може бути корисним для відлагодження або якщо у вас виникають проблеми з апаратним прискоренням.

Installation and Usage Instructions for Candle - GRBL controller application with G-Code visualizer written in Qt
What is Candle?
Candle is an open-source program written in Qt for controlling GRBL controllers. It allows you to visualize the G-code sent to the controller, making it easier to debug and understand the machining process.

Installing required packages:
Update package lists:
Bash

sudo apt update
Install Qt Creator and necessary libraries:
Bash

sudo apt install qtcreator qtbase5-dev qt5-qmake g++ libqt5serialport5 libqt5serialport5-dev
Cloning the Candle repository:
Navigate to the desired directory:
Bash

cd /home/uzfm/Downloads
Clone the repository:
Bash

git clone https://github.com/Denvi/Candle.git
Building and running Candle:
Navigate to the project directory:
Bash

cd rpi-Candle-master/src
Create a Makefile:
Bash

sudo qmake candle.pro
Build the program:
Bash

make
Run the program:
Bash

./Candle
Resolving the black screen issue when visualizing G-code:
Reason: There may be an issue with the graphics backend.
Solution: Install OpenGL ES libraries to enable hardware-accelerated graphics.
Installation command:
Bash

sudo apt install libgles2-mesa-dev
Running the program using software rendering (if hardware acceleration doesn't work):
Bash

LIBGL_ALWAYS_SOFTWARE=1 ./Candle
Additional explanations:
OpenGL ES: This is a standard for 3D graphics, often used in mobile devices and embedded systems. Candle uses OpenGL ES to visualize G-code.
LIBGL_ALWAYS_SOFTWARE=1: This parameter forces the program to use software rendering instead of hardware acceleration. This can be useful for debugging or if you have problems with hardware acceleration.