Denvi/Candle #78: New version of grbl

milezdr, 2016-10-13:
Hi @Denvi  
New version of the Grbl have arrived but in order to use your program you need to update the software do you plan anything about that? 
Here is the link of the new grbl version 1.1  
https://github.com/gnea/grbl/

https://github.com/grbl/grbl/issues/1107

chamnit, 2016-10-14:
@Denvi : Nice GUI! Looks simple and powerful. I'll have to try it out. Let me know if you have any questions on updating your GUI for Grbl v1.1. There's a lot of new stuff and I've written a lot of documentation on what has changed. We're still working out a couple more details but I'll finalize them very soon.

Denvi, 2016-10-15:
I am glad to hear about the new version.

Plans: fix minor bugs > add support for big files (> 1kk lines) > adapt for lasers routers > update for new Grbl 1.1 firmware.

@chamnit Thanks, new Grbl features have to allow to get rid of an array of problems in the GUI programs.

closed 2016-11-27 by milezdr