Denvi/Candle #79: Candle 0.9 not working on Linux Mint/Ubuntu x64

Werwolfx, 2016-10-14:
Hi

i try to run on linux Mint 17.3 but showing error 

./Candle 
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: linuxfb, minimal, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted

Denvi, 2016-10-15:
Hi, please try to run attached script in "Candle/platforms" folder.

[fixdep.zip](https://github.com/Denvi/Candle/files/530756/fixdep.zip)

Werwolfx, 2016-10-15:
This sh not work on my system =(

but i found the issue was indeed that libqxcb.so required libQt5XcbQpa.so.5 and libQt5DBus.so.5 to be present in the same directory.

ldd libqxcb.so 
	linux-gate.so.1 =>  (0xf77bf000)
	libQt5XcbQpa.so.5 => not found
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xf778e000)
	libX11-xcb.so.1 => /usr/lib/i386-linux-gnu/libX11-xcb.so.1 (0xf778a000)
	libXi.so.6 => /usr/lib/i386-linux-gnu/libXi.so.6 (0xf777a000)
	libSM.so.6 => /usr/lib/i386-linux-gnu/libSM.so.6 (0xf7771000)
	libICE.so.6 => /usr/lib/i386-linux-gnu/libICE.so.6 (0xf7757000)
	libxcb.so.1 => /usr/lib/i386-linux-gnu/libxcb.so.1 (0xf7735000)
	libfontconfig.so.1 => /usr/lib/i386-linux-gnu/libfontconfig.so.1 (0xf76f9000)
	libfreetype.so.6 => /usr/lib/i386-linux-gnu/libfreetype.so.6 (0xf7659000)
	libQt5DBus.so.5 => not found
	libgthread-2.0.so.0 => /usr/lib/i386-linux-gnu/libgthread-2.0.so.0 (0xf7656000)
	librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xf764d000)
	libglib-2.0.so.0 => /lib/i386-linux-gnu/libglib-2.0.so.0 (0xf7540000)
	libXrender.so.1 => /usr/lib/i386-linux-gnu/libXrender.so.1 (0xf7535000)
	libXext.so.6 => /usr/lib/i386-linux-gnu/libXext.so.6 (0xf7522000)
	libX11.so.6 => /usr/lib/i386-linux-gnu/libX11.so.6 (0xf73ee000)
	libQt5Gui.so.5 => not found
	libQt5Core.so.5 => not found
	libGL.so.1 => /usr/lib/i386-linux-gnu/mesa/libGL.so.1 (0xf7358000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xf733c000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xf7254000)
	libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xf720e000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xf71f0000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xf7041000)
	/lib/ld-linux.so.2 (0xf77c1000)
	libuuid.so.1 => /lib/i386-linux-gnu/libuuid.so.1 (0xf703b000)
	libXau.so.6 => /usr/lib/i386-linux-gnu/libXau.so.6 (0xf7037000)
	libXdmcp.so.6 => /usr/lib/i386-linux-gnu/libXdmcp.so.6 (0xf7030000)
	libexpat.so.1 => /lib/i386-linux-gnu/libexpat.so.1 (0xf7006000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xf6fec000)
	libpng12.so.0 => /lib/i386-linux-gnu/libpng12.so.0 (0xf6fc4000)
	libpcre.so.3 => /lib/i386-linux-gnu/libpcre.so.3 (0xf6f86000)
	libglapi.so.0 => /usr/lib/i386-linux-gnu/libglapi.so.0 (0xf6f6d000)
	libXdamage.so.1 => /usr/lib/i386-linux-gnu/libXdamage.so.1 (0xf6f68000)
	libXfixes.so.3 => /usr/lib/i386-linux-gnu/libXfixes.so.3 (0xf6f62000)
	libxcb-glx.so.0 => /usr/lib/i386-linux-gnu/libxcb-glx.so.0 (0xf6f4a000)
	libxcb-dri2.so.0 => /usr/lib/i386-linux-gnu/libxcb-dri2.so.0 (0xf6f44000)
	libxcb-dri3.so.0 => /usr/lib/i386-linux-gnu/libxcb-dri3.so.0 (0xf6f40000)
	libxcb-present.so.0 => /usr/lib/i386-linux-gnu/libxcb-present.so.0 (0xf6f3b000)
	libxcb-sync.so.1 => /usr/lib/i386-linux-gnu/libxcb-sync.so.1 (0xf6f34000)
	libxshmfence.so.1 => /usr/lib/i386-linux-gnu/libxshmfence.so.1 (0xf6f31000)
	libXxf86vm.so.1 => /usr/lib/i386-linux-gnu/libXxf86vm.so.1 (0xf6f2b000)
	libdrm.so.2 => /usr/lib/i386-linux-gnu/libdrm.so.2 (0xf6f1b000)

Denvi, 2016-10-15:
> libqxcb.so required libQt5XcbQpa.so.5 and libQt5DBus.so.5 to be present in the same directory

[Candle#79.tar.zip](https://github.com/Denvi/Candle/files/530823/Candle.79.tar.zip)

hablaster, 2016-10-25:
Hi,
I tried with the Candle#79 but I no good result.
Sorry I'm new in linux word.

"This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: linuxfb, minimal, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)"

hablaster, 2016-10-25:
hi,
I have skipped the problem with a simple run/build with QTCreator project on the Github..

Werwolfx, 2016-11-08:
I skipped to and compile from QT.
Close please

closed 2016-11-08 by Werwolfx