Denvi/Candle #84: Windows 10 x64  ERROR:0xc0000005

makemax, 2016-11-20:
Automatically exit

日志名称:          Application
来源:            Application Error
日期:            2016/11/21 0:27:04
事件 ID:         1000
任务类别:          (100)
级别:            错误
关键字:           经典
用户:            暂缺
计算机:           evan
描述:
错误应用程序名称: Candle.exe，版本: 1.0.7.0，时间戳: 0x5831c2ce
错误模块名称: unknown，版本: 0.0.0.0，时间戳: 0x00000000
异常代码: 0xc0000005
错误偏移量: 0x00000000
错误进程 ID: 0x4074
错误应用程序启动时间: 0x01d2434ae78584fb
错误应用程序路径: C:\Candle\Candle.exe
错误模块路径: unknown
报告 ID: f01ec76e-ef02-4fae-b5f4-65f232038802
错误程序包全名: 
错误程序包相对应用程序 ID: 
事件 Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2016-11-20T16:27:04.370452300Z" />
    <EventRecordID>34289</EventRecordID>
    <Channel>Application</Channel>
    <Computer>evan</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Candle.exe</Data>
    <Data>1.0.7.0</Data>
    <Data>5831c2ce</Data>
    <Data>unknown</Data>
    <Data>0.0.0.0</Data>
    <Data>00000000</Data>
    <Data>c0000005</Data>
    <Data>00000000</Data>
    <Data>4074</Data>
    <Data>01d2434ae78584fb</Data>
    <Data>C:\Candle\Candle.exe</Data>
    <Data>unknown</Data>
    <Data>f01ec76e-ef02-4fae-b5f4-65f232038802</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

makemax, 2016-11-20:
Candle_0.9  Can be used normally

Denvi, 2016-11-22:
Can't reproduce. Work fine with Win10 x64.

closed 2016-12-09 by Denvi