Denvi/Candle #86: unstable with pibot controller

fantilator, 2016-11-20:
Hi, 

I'm using the software since grblcontrol 0.7, and have seen that all versions are unstable on windows platforms.
Sometimes the software runs smootly, sometimes, values are not updated, and sometimes the programs just ends with a "This program has ended" messagebox.
My controller is a Pibot with GRBL 0.9G version.
I already tries on several computers, Win 7/Win 10, and all of them are doing the same.  Also tried with my 2nd Pibot controller, and same things happen.

Unfortunately there is no error code poping up.

Denvi, 2016-11-22:
Looks like USB-connection problem, have you tried different cables?

closed 2016-12-11 by Denvi

fantilator, 2017-01-09:
Different cable solved the problem. thank you :)