Denvi/Candle #87: Windows: candle_1.1.zip

Peter6891, 2016-11-22:
Hello!
Have download new version candle_1.1.zip for windows... But it is not run on my operating system (Windows XP SP3)

Denvi, 2016-11-22:
Any error messages? Tested just now on Windows XP SP3 (VirtualBox guest), works fine.

Peter6891, 2016-11-22:
I unzipped the file (candle_1.1.zip), which is downloaded from the website to a folder C:Programm Files\Candle 1.1\
And it isn't run... Message: "Programm does not run!"

Peter6891, 2016-11-22:
(sorry for my English)

Peter6891, 2016-11-22:
may be needed for the program to even some components of the windows were installed?

Denvi, 2016-11-22:
Try [vcredist_x86.exe](https://www.microsoft.com/ru-RU/download/details.aspx?id=40784) please.

Peter6891, 2016-11-22:
installed - did not help. at home more try ...

Peter6891, 2016-11-22:
home installed the program - it works ... (operating system such as windows xp)

starline, 2017-01-03:
добрый день!

У меня тоже наблюдается такая проблема на разных компьютерах под управлением Windows XP 32 Бита

Программа не можем запуститься с вылетом окошка Message: "Programm does not run!"

Peter6891, 2017-01-03:
на следующий день программа запускается и нормально работает(на тех компьютерах, на которые устанавливалась но сразу не заработала, а вылетала с окном "Программа не запускается"). причина мне не известна. возможно что это проблема с самой операционной системой - может позже что-то обновляется и после перезапуска в определённый момент запускается...

starline, 2017-01-03:
Хорошо бы разобраться почему так
На другом компьютере с такой же операционной словил ошибку при запуске.
"Ошибка при инициализации приложения 0xс00000 1d"

starline, 2017-01-03:
Отвечаю сам на свой вопрос.
По ошибке нашел следующую информацию 

> Проблема в том, что процессор на данном ПК не поддерживает SSE2 - http://en.wikipedia.org/wiki/SSE2

Peter6891, 2017-01-05:
я же говорю - установил программу - не запускалась. оставил установленную. на следующий день уже не переустанавливал - она со вчера установлена была. просто запустил - запустилась. при дальнейших запусках всегда запускалась. и так на трёх компах.

Denvi, 2017-01-25:
> как это можно исправить? Ну кроме как заменить железо )

Пересобрать Qt без поддержки SSE2. Если это вообще реально.

closed 2017-01-25 by Denvi

starline, 2017-10-03:
Еще дополню данную тему. Проблему на некоторых машинах можно решить. Установить драйвер с поддержкой SSE2.

Peter6891, 2017-10-03:
а где его взять? (и драйвер чего? - CH340G ?)