Denvi/Candle #89: Display

Peter6891, 2016-11-26:
whether it is possible to make a display indicating the status of the machine - for example, the state of the end position sensors, emergency stop button, etc.?

Denvi, 2016-11-27:
XYZ-axes limits, Probe, Door, Hold, soft-Reset, cycle Start pins states will be added to output string in top-left corner of visualizer window.

Peter6891, 2016-11-27:
it is already in some versions of the program or is it planned to do?

Denvi, 2016-11-28:
Added to 1.1.4:
F/S - feed/spindle;
PS - pins state;
AS - accessory state.

[image]

Peter6891, 2016-11-29:
and add in the option settings so you can change the polarity of the input and output terminals in the program as possible?

Denvi, 2016-11-29:
Has answered in #83.

Peter6891, 2017-09-01:
Бодрый день... Есть проблема, не знаю в чём причина - не могли бы подсказать?
Запускаю программу, плата подключена, прошивка контроллера 0.9j или 1.1f (разные пробовал).
В строке "Статус": надпись "Готов", но периодически мигает красным и надпись "Неизвестно"...

куда копать?

Peter6891, 2017-09-01:
есть видео, но прикрепить тут вроде нельзя к письму...

Peter6891, 2017-09-01:
[image]
[bandicam 2017-09-01 17-07-49-218.zip](https://github.com/Denvi/Candle/files/1270554/bandicam.2017-09-01.17-07-49-218.zip)

Peter6891, 2017-09-01:
прикрепил архивом выше...

closed 2025-09-15 by github-actions[bot]