Denvi/Candle #90: Version 1.1.1

frobgm, 2016-11-26:
Hello Denvi
I tried yours new version, 
the presets the older 0.9 version is better than new version 1.1 and 1.1.1 

and windows 1.1.1 version doesn't show opened g-code drawing
linux 1.1.1 version drawing good

Robert

Denvi, 2016-11-26:
Hi, what presets you are talking about?

> upload english/english translation file

Version 1.1 is pre-release, will upload close to release.

frobgm, 2016-11-26:
[image]
the new 1.1.1 version program jog presets
[image]
and the older version 0.9, I think better than ver1.1.1  jog presets

Denvi, 2016-11-26:
Old version has presets for jog "Step", new one - "Feed".  Last one works with new jog function of Grbl v1.1, where you can continuously move tool in selected direction by holding cursor buttons/keys.
May be you want other feed values?

frobgm, 2016-11-26:
:( not so good, I used some big cnc machine, all of them machines jog function using predefined distance
0.01 , 0.1, 1, 10, 100 mm 
for example therefore very easy setting the work null point with calibrated steel piece, without electronic  null probe 

it could be two row button? first feed rate, second step distance

Denvi, 2016-11-26:
Will think about it.

frobgm, 2016-11-26:
Thanks! I hope :)

Denvi, 2016-11-27:
Buttons replaced with 2 comboboxes, "Stop"-button added.

Numpad bindings:
- \<\\> - spindle speed decrease;
- \<*\> - spindle speed increase;
- \<7\> - previous step;
- \<1\> - next step;
- \<-\> - previous feed;
- \<+\> - next feed.

[Candle_1.1.3.zip](https://github.com/Denvi/Candle/releases/download/v1.1/Candle_1.1.3.zip)

frobgm, 2016-11-27:
Hello Denvi
Thanks!

frobgm, 2016-11-27:
one thing, since version 1.1 on virtulbox windows 7 doesn't show nothing the drawing window , I usualy tested first there...
not big problem, just tell

Denvi, 2016-11-27:
Have you installed Guest Additions & turned on "3D acceleration" in machine options?

frobgm, 2016-11-27:
No, so far doesn't need it, the older version working well without "3D acceleration"...
but not big problem, I tested another pc...or linux version :)

Denvi, 2016-11-27:
There is some features with OpenGL realization among different versions of Qt, thus for 5.4.2 we need 3d acceleration option for virtual box guest.

frobgm, 2016-11-27:
I understand!
Thank you for your work!

frobgm, 2016-11-27:
I read some post yours program how to run raspberry pi 3, so I little bit lost, you have a simple method how I run yours program on raspberry?
You try on raspberry? how fast ? ( youtube vid ?)

Denvi, 2016-11-28:
I haven't run program on Raspberry 3, and think there is no simple method to do it.

frobgm, 2016-11-28:
Dear Denvi
I try to using ver 1.1.4 new 2 combo boxes , not so good...
feed rate not to important, but to you need it could be slider?
and the distance , the oldiest buttons, one row, four or five buttons?
easier to choose I think...

Denvi, 2016-11-28:
Buttons works well only with one parameter (feed or speed) not both (to many buttons to select) and it isn't expansible (can't add more presets).
Sliders doesn't show values (user can't see what he selects). Doesn't works well with discrete nonuniform values.
So far I will leave as is, at least until it's a better idea.

frobgm, 2016-11-28:
I understand and you're right.
and I can't think of a better now...
Perhaps the command state response window it could be smaller, and some button bottom or between two window ( showing and command window)  , just I thinking "loudly" :)
I try the Scroll Lock button, it is great idea!

Which Qt version do you use ?

Denvi, 2016-11-30:
All recent builds on Qt 5.4.2.
5.5 - problems with deployment on linux platform.
5.7 - only 64-bit version for linux.

frobgm, 2016-11-30:
Thank you for your answer!

frobgm, 2016-12-03:
Hello Denvi
I played a bit picture manipulating sw, look this :)
[image]

too much sliding on the right side, I think on the bottom place from may be more useful things, and the g-code window there could be a new tab 

I think from usable your program...
What do you think?

Denvi, 2016-12-03:
Yes, some panels need to be removed from right side. 
So far I adhere to the following idea: file manipulation - left side, cnc control - right side.
Program/console - dock widgets, can be placed on bottom/left/right side of visualizer, including one place with tabs.

[image]
[image]
[image]

But it will be made not soon.

frobgm, 2016-12-03:
Why important for you the program and the console window?
I think right side  buttons and features important looking all times, and get without slide.
and the screens , usually the screens have 16:9 ratio, now yours program version not to friendly this ratio :)
I think you look for it after...

have nice day!

frobgm, 2016-12-03:
I thinking all day what is the simple solution.
one idea the right side buttons and features it could be to columns, what do you think?

Denvi, 2016-12-03:
Anyway i think columns layouts are more flexible than rows, look for "Blender" old versions & latest.

frobgm, 2016-12-03:
I looked it up, 
see the picture
[image]

all buttons have a right side, not both , but if you thinking left-handed people, one tick on the settings page left or right handed.

I think the people like see the drawing window, and doesn't like through out his hand the drawing window
on my opinion one side button heap and bottom, this is handy

 yours program, but I say my opinion :)

Denvi, 2016-12-03:
All things are good in moderation.

[image]

> I think the people like see the drawing window, and doesn't like through out his hand the drawing window
on my opinion one side button heap and bottom, this is handy

Agree with you, but some things can be placed everywhere.

frobgm, 2016-12-03:
:)
not to important things really placed everywhere...
but important operation buttons are usually one place
that your  program is it, but opening little windows and scrolling .... this should be changed

frobgm, 2016-12-04:
Hello Denvi
I found same which like it
http://wiki.linuxcnc.org/uploads/gmoccapy_4_axis.png
I think the layout is good.... what do you think?
but this program layout made 5:4 screen, which is becoming increasingly rare
16:9 screen ratio has all laptops and many desktop computer...but you must know :)

and your program screen colouring is better than me as gmoccapy screen

frobgm, 2016-12-04:
even an idea guide  :)
http://www.polgardidesign.hu/images/stories/UCCNC.png
over many tabs, on the left the blue streak is the hidden jog buttons...

Denvi, 2016-12-11:
The user interface will change over time.

closed 2016-12-11 by Denvi