Denvi/Candle #91: 8376709 "Send from current line"-command added

milezdr, 2016-11-27:
This is more a question not a issue but how I can use this   I try to right click on the g code line a need to start with and it appears only insert line or delete line
Thank anyway :)

Denvi, 2016-11-27:
Hi, you should select line, then press & hold "Send" button, wait for menu popup, click on "Send from current line" menu item.

milezdr, 2016-11-27:
Thanks I will try this tomorrow :)

milezdr, 2016-11-28:
sorry I have one more question, Can I use candle 1.1,4 with grbl 0.9j or I need to flash new grbl 1.1 
Because I don't have latest post processor in my CAM software for grbl 1.1 don't know if need some changes in post processor for the new grbl 1.1 version? Any thoughts?

Denvi, 2016-11-29:
As for "Send from current line"-feature, it will be added to 1.0 version and would work with Grbl 0.9.

closed 2016-11-30 by milezdr