Denvi/Candle #93: simulation

Peter6891, 2016-11-29:
and simulation mode processing and rendering the trajectory milling cutters motion only works when the connected circuit board in the machine?
It is impossible to do so and could not connect to the board to check the control code for the machine and view the trajectory of movement of milling cutters?

Denvi, 2016-11-29:
There is no simulation function and won't be in Candle. Only check through firmware.

Peter6891, 2016-11-29:
[image]

Peter6891, 2016-11-29:
there are red circled "test mode" - this is what test? Is it not a simulation control program ??

Denvi, 2016-12-11:
No. It is "Check mode" = "$C" - command, described in GRBL firmware docs.

closed 2016-12-11 by Denvi