Denvi/Candle #94: Software Crashing

balajisah, 2016-11-30:
Hi, 
I use a GRBL Controlled mini CNC mill purchased from Aliexpress.
It came with GRBL and Candle software 0.9
I'm facing problems with the candle software
1. candle 0.9 software simulation freezes when running a program and hangs.
2. Candle 1.0.X version also crash during the program run.

Do i need to upgrade the firmware on the GRBL cnc Board in machine?

Denvi, 2016-11-30:
I have zero information about your cnc, controller, firmware, program you trying to send, conditions before Candle hangs. Сan't answer something responsible.

What is "software simulation"?
Do you have similar problems with other g-code senders?

balajisah, 2016-12-03:
Hi Denvi,
Thanks for your reply.
I'm sending the photo of the controller, machine, & the cnc program.
[redenvicandlesoftwarecrashing94.zip](https://github.com/Denvi/Candle/files/628948/redenvicandlesoftwarecrashing94.zip)
I'm facing below problems.

1.       Machine Axis Jog functions are working fine only with candle 0.9 version. If I run any program The program visualization stops and software gets hanged. Machine stops to move or moves to hit the limits.
2.       I used new versions, in that the Axis is not moving in jog.
3.       When i again work on 0.9 version, Return to origin command is not moving the tool tip to the work offset position.
Please help with solution.

Denvi, 2016-12-03:
> Do you have similar problems with other g-code senders?

Have you tried "Universal G-code sender"?

userjim, 2016-12-09:
Maybe I can help with providing info on the error I get. I don't know if it is what Balajisah gets.

I just started playing with a small tabletop 3-axis CNC. I used Easel to create the g-code, downloaded it and opened it in version grblcontrol 0.9. The example is a simple gear. I uploaded the nc file and a text file I made from text in the console (which includes the error). It got about 1/2 way through the process when it froze. I was cutting into plywood with a brand new 1/8" upcut bit. 

By the way, I also got the same error when etching a name into aluminum with an etching bit.
I think the issue is related to the CNC. The same file runs fine if not cutting into anything. So, I think it has something to do with what happens if the material gets harder in the middle of the cut and slows it down a tad. Maybe it gets confused or doesn't reply in time or something.

The CNC I have is a DIY model that is sold in Ebay, Amazon, and I think AliExpress. Search Ebay for: Mini 3-Axis CNC 1610 Router Engraver PCB PVC Milling Wood Carving Machine DIY

I had to assemble it. The controller is not an actual Arduino board, but is running Arduino with some version of Grbl. I assume the latest, but I don't know how to find out. It uses a 24V power supply and a mini-usb port to the PC. The board must be made specifically to run a CNC. It also has a second power slot to run a laser instead of a spindle.

I hope this helps. I'm going to try a slower feed rate, I already lowered it from what Easel wanted to use. Maybe I will make the cutting depth smaller too. I just didn't want to have grand-kids before it finished cutting.

Jim

[Grbl error.txt](https://github.com/Denvi/Candle/files/641055/Grbl.error.txt)
[Gear.nc.txt](https://github.com/Denvi/Candle/files/641057/Gear.nc.txt)

userjim, 2016-12-09:
Nope, faster spindle speed, shallower cut depth (0.5 mm) and slower movement speed. It had 12 passes to do. After it caught up to the spot where it died last time (about 6 passes), it made it 1 more round and died again. same errors.

I am looking in the settings and there is something about Rapid speed: 500. I don't know what this does. The help ? in the corner of the dialog is not working. I also don't know what the Feed setting does.

BTW, I am using Windows 10.

Denvi, 2016-12-11:
Everything that you have described, is problems with an electronic part of your CNC.
Controller works unstable.

darkowls, 2016-12-17:
I am also having issue with software failing to start on Windows 10 - it starts once, I can choose COM port, and control my CNC. But after closing - I just cannot open software again. 

Can't find any log file to narrow the problem. Can you help?

felixa4, 2016-12-31:
hello Denvi, I can not assemble the sw: I tried with 5.4.2.mingw qt and qt 5.7.1 mingw; it assembles sw, generates a 40 MB file around, then "the Candle prog stopped working ....".
help me please, bye
happy new year to all

closed 2017-01-17 by Denvi