Denvi/Candle #95: save coordinates and locating

ASHIT-AXAR, 2016-11-30:
Hello dear Denvi.
this is a fine, smooth, and fast software. i really like it.
(i am using v0.9 currently)
----------------------------------------------------------------------------------------------------------------------------------
is it possible to add a coordinate saving to the software.
something like saving coordinates (both of them) to a (predefined or known) file (which can be accessed later) automatically (not asking or showing a message or prompt to do that) when pressing the 'Send' button!
this will be really useful when there are mistakes and failures in the operation. (to start over the job)
------------------------------------------------------------------------------------------------------------------------------------
also i prefer that jogging buttons to use G1 command instead of G0 to position the machine on the job.
(moving by current speed of the machine)
is it possible to add an option in settings to switch into this mode ?
------------------------------------------------------------------------------------------------------------------------------------
and if it could be define alternative control shortcut keys (for X+, X-, Y+, Y-, Z+, Z-)
------------------------------------------------------------------------------------------------------------------------------------

thank you for your great works !

Denvi, 2016-12-03:
Hello

> may you add a coordinate saving to the software?

"Zero XY" & "Zero Z" command buttons saves the offsets from home location to "settings.ini" file in the following fields:

~~~
storedX=63.79
storedY=37.2
storedZ=10
~~~

When you press "Restore XYZ"-button all offsets restored, even if the "Candle" has been restarted.
So your work coordinates saved/restored already or if other function can be necessary to you - describe in more detail.

ASHIT-AXAR, 2016-12-04:
sorry for my late reply, i was changing the OS to win10. (and it took a while getting things set up)
(i got in trouble with win10 and i'm going to back to win7)
''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''

> When you press "Restore XYZ"-button all offsets restored
i never touched that button before.
but it doesn't works good in my case
i think i have configured grbl board not so well but i am using it this way for a month
(my zero [X, Y] is [left, near] at [-1199, -799] mm)
when i push "Restore XYZ" it gets more negative and 
i would try another configuration as soon as i know what is the best 

i download new Candle 1.4 (new changes are really fine, i like it)
jog buttons take no action and nothing visible in visualizer. but user commands works well.
( i realized these all problems are with Win10 - grblControl 0.8.4 was very good in win7)

Denvi, 2016-12-05:
Try to work in local coordinates, use "Zero XY" & "Zero Z" commands.

Candle 1.1+ is for Grbl 1.1 firmware version. Firmware should be configured at least with command:
~~~
$10=1
~~~

Also program should work good with Win 10, i don't think it is windows relative problems.

closed 2016-12-11 by Denvi