Denvi/Candle #96: Candle_1.0.9 correction parameter for each axis

RadioGG, 2016-12-04:
Hello Denvi,
Your soft Candle v1.0.9 is very fine and it's very easy to start. 
For the moment i have just this basic problème: 
- i cannot get a unit displayed for a unit moving on each axis. Could you tell me where the distance correction parameter for each axis is?
Thank you very much for your help.
Cordially.
Gérard

Denvi, 2016-12-05:
Hi, here is firmware settings manual: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

> $100, $101 and $102 – [X,Y,Z] steps/mm

>Grbl needs to know how far each step will take the tool in reality. To calculate steps/mm for an axis of your machine you need to know:

>    The mm traveled per revolution of your stepper motor. This is dependent on your belt drive gears or lead screw pitch.
 >   The full steps per revolution of your steppers (typically 200)
  >  The microsteps per step of your controller (typically 1, 2, 4, 8, or 16). Tip: Using high microstep values >(e.g., 16) can reduce your stepper motor torque, so use the lowest that gives you the desired axis >resolution and comfortable running properties.

>The steps/mm can then be calculated like this: steps_per_mm = >(steps_per_revolution*microsteps)/mm_per_rev

>Compute this value for every axis and write these settings to Grbl.

Parser commands for units select:

~~~
G20 - imperial
G21 - metric
~~~

RadioGG, 2016-12-05:
Hi Denvi, 
I'm sorry, I had not seen your firmware settings manual. Thank you again for your quick reply. 
Have a good day.

RadioGG, 2016-12-05:
Hi dear Denvi,
I have the v1.0.9 and this manual is for the v0.9. They are  not exactly the same parameters. However i will try.
Perhaps a new help for v1.0.9 in your projects?

closed 2017-01-06 by Denvi