Denvi/Candle #98: How to change lenguage/translate?

Konesuke, 2016-12-12:
Can't find the option in the settings

Denvi, 2016-12-28:
Hi, there is no language setting in "Candle", it is selected by system regional settings.

fedya, 2017-02-20:
I'm fixed it

https://github.com/Denvi/Candle/issues/121

misisl, 2017-02-23:
hello,
i have regional system (Win7_64bit) setting as Czech (fomat, location and keyboard), but Candle is in English language.  Where is problem?  Can you add option to setting.ini for overwrite system language setting?  

Thank you!

fedya, 2017-02-23:
@misisl 
There is only such translations
```
/usr/share/candle/translations/candle_es.qm
/usr/share/candle/translations/candle_fr.qm
/usr/share/candle/translations/candle_pt.qm
/usr/share/candle/translations/candle_ru.qm
```
No Czech language there, you need to prepare a patch with proper translation.
If you want to do it look at

`https://github.com/Denvi/Candle/blob/master/src/translations/candle_ru.ts`

misisl, 2017-02-23:
ou, sorry.
 I found in "translations" subdir file "qt_cs.qm" with czech strings and think ...   :)

i will prepare "candle_cs.qm" file.

fedya, 2017-02-23:
@misisl 
Make sure that you see Candle_cs.qm and not qt_cs.qm
If it is point me where to download such file.

misisl, 2017-02-25:
hi, can you compile test version for me, please?

[translation RU-CZ_v3.zip](https://github.com/Denvi/Candle/files/801495/translation.RU-CZ_v3.zip)

misisl, 2017-04-13:
hi, i have translated files to czech language.  Can you add it to project?

[Candle_cs.zip](https://github.com/Denvi/Candle/files/919988/Candle_cs.zip)

fedya, 2017-04-13:
Probably project abandoned?

EBEALIEN, 2019-03-07:
good morning
I dare: is it possible to set the program in Italian? How can I do?
Thank you very much

sandeemontonpalin, 2020-08-17:
what's the camera for?

closed 2025-09-14 by github-actions[bot]