Denvi/Candle #99: Visualization is not working on Samsung n130

ghostbuster2, 2016-12-21:
Do not work on Windows 10/8.1/7
Work fine on ubuntu 16.10
Here is screenshots https://yadi.sk/d/zpQ9roQw34Ec9Y

ghostbuster2, 2016-12-25:
В нетбуке установлена встроенная видеокарта семейства Mobile Intel(R) 945 Express Chipset. В версии программы из поста #64 визуализация завелась, но хотелось бы все-таки рабочую последнюю версию. Спасибо!

ghostbuster2, 2016-12-26:
0.9. версия работает, подозреваю потому-что в ней есть все эти методы рендеринга
"Visualizer works with OpenGL/Angle/Software renderer"
Нужна по возможности версия 1.1 т.к. в контроллер залита последняя прошивка grbl

Denvi, 2016-12-26:
Нужно собирать на Qt 5.5, на днях сделаю.

ghostbuster2, 2016-12-26:
Ок, спасибо!

Denvi, 2016-12-27:
[Candle_1.1.7_.99.zip](https://github.com/Denvi/Candle/releases/download/v1.1/Candle_1.1.7_.99.zip)

ghostbuster2, 2016-12-27:
Благодарю за оперативность, все работает.

Denvi, 2016-12-28:
> Do not work on Windows 10/8.1/7

Про XP речи не было... придется жить с ошибкой.

ghostbuster2, 2016-12-28:
Если на функционал не влияет, то можно жить,😀

Denvi, 2016-12-28:
Теперь не должен сыпать ошибки, если [редиску](https://www.microsoft.com/ru-ru/download/details.aspx?id=30679) установить.

[Candle_1.1.7.zip](https://github.com/Denvi/Candle/releases/download/v1.1/Candle_1.1.7.zip)

ghostbuster2, 2016-12-28:
Значит сейчас расклад такой:
До установки "Распространяемый пакет Visual C++ для Visual Studio 2012 Обновление 4" по вашей ссылке выдавало такую ошибку
[image]
После установки - ситуация не изменилась, ошибка все та же.
Потом решил установить этот [пакет](https://www.microsoft.com/RU-ru/download/details.aspx?id=40784) После его установки ошибка сменилась на такую
[image]

Denvi, 2016-12-28:
Ясно, буду искать тачку на древнем чипсете, но что-то мне подсказывает - бесполезно возиться с XP.

ghostbuster2, 2016-12-28:
Вот и я к этому склоняюсь, верну наверное восьмерку и горя знать не буду. Спасибо за приложенные усилия.

prulitarit, 2017-01-23:
> Ясно, буду искать тачку на древнем чипсете, но что-то мне подсказывает - бесполезно возиться с XP.

I successfully reproduced error, ghostbuster2 talking about,  in the virtualbox windows xp:

Without any vcredist installed:  - msvcp120.dll and msvcr120.dll not found error.
With vcredir 2013 installed - strnlen not found in msvcr.dll error

if I remove d3dcompiler_47.dll file, strnlen error goes away.

you can use virtualbox. it can emulate old chipsets.

Denvi, 2017-01-24:
Yes, i have reproduced this error on virtualbox (but i can't simulate OpenGL 1.0 on it while my card was 4+) and on hardware. Anyway WindowsXP doesn't works with ANGLE, you should install newer graphics card.

prulitarit, 2017-01-24:
I can not get newest graphics card because only Intel 945GM internal video chipset in this laptop. 

> Graphics card with OpenGL 2.0 support

My latest OpenGL version is 1.4 and Candle seems to work fine, (exept this error at startup). 
How can I determine actual rendering mode (OpenGL/Angle/Software renderer) ?
And what difference between these modes?