Denvi/Candle #104: (maybe) a better "Port Opening Method"

jiarko, 2017-01-01:
Hi,  I was trying to build up "my grbl interface" using Qt and met you wondeful build!!!

I've been through several other grbl gui and personally struggled to get the grbl message upon port opening "**Grbl 0.9j ['$' for help]**"; using Qt the best (maybe the only) way to accomplish this without resetting is to add the following call:

m_serialPort.setDataTerminalReady(true);  

by doing this, you can see on arduino Uno the "L" led flashing and the the "RX" and TX" too.... afterwards, you get the grbl message output without any need to "send a reset command. Same happens when connecting via UGS b ut couldn'f find how this is done there....

thought this can be added to the "openPort()" method.

hope this helps in some extent.

ciao

Denvi, 2017-01-02:
Hello.

DTR-line manipulation is a very tricky with Qt, and depends on platform.
For example, in Linux I can't get open port procedure without short pulse on DTR line.

Besides, in most cases hard reseting of the controller (DTR - is a hard reset for arduinos) means zeroing of machine coordinates that is inadmissible at restart of the program. In this case I advise to disconnect DTR line from RESET pin of MCU.

jiarko, 2017-01-02:
Hi,

testing on my pc DTR didn't reset the arduino coords. But I'm not an expert, I've just realized that when connecting UGS I see the leds on arduino flash and the I get the "welcome message", in Qt if don't do the DTR, even with a "reset command" I can't get the same behavior. 

anyhow, **very good job!!** I'm new on github activities, but I'd like to give my help in the application you wrote, I'm a cnc user and I've been using some cnc interfaces (some of them semi-professional), maybe I can give my contribution in GUI... 

let me know if it is ok for you and how I can do it.

ciao

felixa4, 2017-01-02:
hi jiarko, can you help me to assemble this source code trough QT?
thanks, bye

jiarko, 2017-01-02:
do you men the "m_serialPort.setDataTerminalReady(true);" ?

felixa4, 2017-01-02:
No, entire candle source code. I try to assemble , qt generates candle.exe for 40 mb. After this the exe hangs...
Help me,please

felixa4, 2017-01-02:
Yesterday i have opened an issue but anyone have answered

closed 2017-01-06 by Denvi