Denvi/Candle #105: grbl 1.1 GUI not working properly

jiarko, 2017-01-04:
Hi,

I've tested the GRBL 1.1 and realized that:

1. tool position on path is not updated (seems the toll is waiting in its start position even if from table I read the command sent updating)

2. in the "State" group, the subgroup "machine coordinates" shows the FS into the "Z" textline but in this way it is not completely shown becuse too many characters to be shown

3. when in "idle" it is shown as "unknown"...

do I miss some settings to let the GUI understand it is a different GRBL version?

thanx
Andrea

Denvi, 2017-01-04:
Hello, download Candle 1.1.x is for GRBL 1.1.

Denvi, 2017-01-04:
It's right one & should work with GRBL 1.1.

jiarko, 2017-01-04:
Sorry, I did wrong. The cande_1.1.7.zip has the"exe" file that works properly,

If I download the source code it is not updated yet... into the "candle.pro" file the VERSION is set equal to 1.0.11.. and once compiled does not work properly... can you make available the source code too?

thanx

Denvi, 2017-01-04:
Checkout "grbl_1_1" branch.

jiarko, 2017-01-04:
Thanx

closed 2017-01-06 by Denvi