Denvi/Candle #108: Visualization issue

milezdr, 2017-01-11:
When I open G code the Visualization work but when I press send it disappears.. While doing that if I go to settings and press just ok visualisation shows up again

Denvi, 2017-01-17:
No video.

milezdr, 2017-01-17:
[https://www.youtube.com/watch?v=lWnF_oUAATA](url)

Denvi, 2017-01-17:
The same behavior for all g-code programs?
Has "Geometry simplify"-option any sense on the problem?
"Reset"-button (to the right of "Send") or "Send from current line"-command cause the same?
Does your graphics card support OpenGL 2.0+?

Offtop: steps per rev aka $100?

milezdr, 2017-01-17:
Yes, for all g code's 
I have activated the Geometry simplify but still the same 
The Reset and Send from current line I don't know  I'll test  but I think its the same 
OpenGL 4.4 (AMD Radeon R5 M330 with 293 ext.)

And for steps per mm I have  $100=20     I use 1/4 microsteping with GT2 belt and 20 tooth pulley count

milezdr, 2017-01-23:
anything new about my problem?

Denvi, 2017-01-23:
> The Reset and Send from current line I don't know I'll test but I think its the same

Waiting...

ngenm, 2019-02-02:
The same problem. I don't understand what to check?

GPU Caps Viewer v1.41.3.0 works fine including all demos.
[gpu_caps_viewer_report_intel_hd_4400.txt](https://github.com/Denvi/Candle/files/2824900/gpu_caps_viewer_report_intel_hd_4400.txt)

closed 2025-09-14 by github-actions[bot]