Denvi/Candle #109: Jog Function

ChinaCNC, 2017-01-17:
The Jog Function is no longer working on my Chinese CNC since 1.1.6 and also now on 1.1.7

Was working on all previous versions.

Windows 10 64.

Spindle speed g code visible  in the window, and working fine. Jog is dead, also from keyboard.

Denvi, 2017-01-17:
I have added 1.1.5 version to releases.
Show how jog works in 1.1.5 and doesn't on 1.1.7.

ChinaCNC, 2017-01-17:
Jog works fine on 1.1.5, but 1.1.5 does not have laser support :-)

Denvi, 2017-01-17:
Laser support was added in 1.0.1/1.1.1.
Post screenshots/g-code programs/program settings/firmware settings with your problems.

ChinaCNC, 2017-01-18:
I do not have all those details, machine is same as https://www.aliexpress.com/item/CNC-1310-diy-cnc-engraving-machine-mini-PVC-PCB-Milling-Machine-Wood-Carving-machine-cnc-router/32719986096.html?spm=2114.30010308.3.70.OKEdHj&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_10000009_10084_10000025_10083_10080_10000029_10082_10081_10000028_10110_10111_10060_10112_10113_10062_10114_10115_10056_10055_10037_10054_10033_10059_10032_10099_10078_10079_10000022_10077_10000012_10103_10073_10102_10000015_10096_10000018_10000019_10052_10053_10107_10050_10106_10051-10102,searchweb201603_2,afswitch_3_afChannel,single_sort_1_default&btsid=a83a498a-d086-4ab7-84a1-5c5ef3f49223

closed 2017-01-25 by Denvi