Denvi/Candle #110: Кнопки перемещения

DrGrbl, 2017-01-20:
В версии 1.1.7 не работают кнопки перемещения в ручном режиме. В версии 1.0.11 работает, но на визуализации нет перемещения шпинделя, постоянно стоит в 0X 0Y, в версии 1.1.7 нормально движется по траектории.  В чем проблема? И не работает кнопка обнуления координат, ни в одной версии.

Denvi, 2017-01-22:
Приветствую.
Пару снимков экрана скиньте с обеими версиями. Покажите, что и как у вас не работает.
Какая прошивка используется?
Вывод команды "$$" также необходим.

DrGrbl, 2017-01-22:
С визуализацией и кнопкой обнуления разобрался, но так и не работает ручное перемещение в версии 1.1.5-1.1.7. В версиях 1.0.11 и ниже все работает.

Denvi, 2017-01-25:
Candle версий 1.0.х для прошивок GRBL 0.9 и ниже, 1.1.х - GRBL 1.1.
Обновите прошивку.

SadovovAlex, 2017-01-29:
Так же, не работает перемещение по кнопкам ручного перемещения в программе v1.1.7
[CTRL+X] < Grbl 1.1e ['$' for help]
S300 < ok
$$ < $0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=400.000
$101=400.000
$102=700.000
$110=800.000
$111=800.000
$112=500.000
$120=200.000
$121=200.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

Denvi, 2017-01-29:
Не вижу криминала в настройках прошивки. Как насчет "settings.ini" файла?

SadovovAlex, 2017-01-29:
ощущение что кнопка ничего не выполняет =)
Должна ли писаться команда перемещения в консоль при этом? у меня не пишется ничего.

Переименовал в txt
[settings.txt](https://github.com/Denvi/Candle/files/737571/settings.txt)

Denvi, 2017-01-29:
Команда не должна выводиться в консоль.
Задайте шаг и подачу (скорость перемещения), например как на изображении ниже (судя по файлу настроек, сейчас не задана подача).

[image]

SadovovAlex, 2017-01-29:
В программе заданы! 
Какие настройки в файле? может проблема в том что в файл не переносятся настройки?
И еще заметил что последняя цифра Буфер растет при нажатии
[image]

Denvi, 2017-01-29:
> Какие настройки в файле? может проблема в том что в файл не переносятся настройки?

~~~
jogFeed=
~~~

Настройки восстанавливаются при перезагрузке программы?

Буфер:
- 1 число - количество байт записанных в буфер последовательно порта;
- 2 - количество команд в буфере, на которые еще не пришел ответ от контроллера;
- 3 - количество команд в очереди.

Если растет последнее число - на команды перемещения нет ответа от контроллера, либо команды посылаются слишком часто.

В архиве версия с выводом команд перемещения в консоль (в настройках необходимо включить флаг "Отображать команды интерфейса пользователя").

https://github.com/Denvi/Candle/releases/download/v1.1/Candle_iss110.tar.gz

Протестируйте и отпишитесь с результатом вывода в консоль. Спасибо.

SadovovAlex, 2017-02-03:
Сорри что долго не отвечал.
Протестировал новую сборку, пишет некую ошибку 22

[CTRL+X] < Grbl 1.1e ['$' for help]
$J=G21G91X1Y0Z0F0 < error:22
$J=G21G91X-1Y0Z0F0 < error:22
$J=G21G91X0Y1Z0F0 < error:22
$J=G21G91X0Y-1Z0F0 < error:22

Denvi, 2017-02-03:
Судя по всему, у вас зависает контроллер. На оси Z шагов на мм больше почти в два раза, чем на X и Y.
Думаю нужно увеличить время опроса контроллера (15 мс - очень частый опрос), либо ковырять параметры прошивки, что уже вне моей компетенции.
Хотя мой тестовый контроллер на Arduino Uno прекрасно работает с вашими настройками.
Попробуйте вручную отправлять команды $J с движением по оси Z (в том числе и напрямую через терминал, "pytty", к примеру), посмотрите что получится.

SadovovAlex, 2017-02-04:
вы были абсолютно правы, проблема оказалась в настройках GRBL, почему-то ускорение по оси Z было выставлено в  0 ! =(
$122 = 200.000    (Z-axis acceleration, mm/sec^2)

Спасибо.
Собственно пожелание, можете ли Вы добавить проверку параметров прошивки $120/$121/$122 на не нулевое значение.

Вывод команд в консоль при перемещении на мой взгляд полезен, добавьте в будущие версии или опцию такую.

И поле консоль сделать в самом низу, чтобы оно было шире.
И вывод параметров прошивки примерно так, как в UGS, c расшифровкой
Grbl 1.1e ['$' for help]
>>> $$
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 700.000    (Z-axis travel resolution, step/mm)
$110 = 800.000    (X-axis maximum rate, mm/min)
$111 = 800.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 200.000    (X-axis acceleration, mm/sec^2)
$121 = 200.000    (Y-axis acceleration, mm/sec^2)
$122 = 200.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok

Lenivets71, 2017-02-08:
Подскажите пожалуйста. программа Candle версия 1.1.7 прошивка 1.1 .Версия драйвера CH340 3.4.2014.8 от 8.8.2014г и там и там Виндовс XP сервис пак 3 дело в чём, на ноутбуке запускается нормально все функции работают исправно а на стационарном компе не работает нормально перемещение , тоесть задаёшь шаг непрерывно, подача 500 и станок начинает дёргаться . А на ноуте такого нет идет глатко.

Lenivets71, 2017-02-09:
любое перемещение по осям, а если ставлю шаг фиксированный (0.01, 0.1, 1, 5, 10, 100) выполняет нормально.

grblkit, 2017-09-15:
Точно такая же проблема,  на 0.8 работает,  сразу пробую на 1.1.7 не работают кнопки перемещения

p1pengod, 2017-09-16:
На прошивке grbl_v1.1f.20170801 все работает. Проверьте свою прошивку

grblkit, 2017-09-16:
Да,  вчера нашел в чем проблема,  чтобы обновить нужно чип выпаять и программатором обновить или можно через USB,  у меня китайская плат woodpecker grbl 0.9

caha200762, 2018-01-20:
Здравствуйте!
Помогите разобраться.
Прошивка grbl_v1.1f.20170801, программа Candle_1.1.7 не работает кнопка дамой,

Denvi, 2018-01-21:
Приветствую.
С $22 по $27 все настроено?
Попробуйте включить в настройках "Candle" отображение команд в консоли и посмотреть код ошибки, выводимой в консоль, после нажатия на кнопку "Домой".

caha200762, 2018-01-21:
Приветствую.
$20=1
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=25
$27=1.000
error:9

Artcl, 2018-02-08:
Добрый день. При загрузки уп. Возникает ошибка Т1Мб<error:20, но при нажатии пропустить все нормально работает, что это может быть и счем связано. Эта же уп в других программах грузится нормально без ошибок.

pipitos11, 2018-02-24:
Добрый день, использую canddle 1.1.7 grbl 1.1 через раз работает коррекция скорости. В чем может быть проблема?

caha200762, 2018-02-28:
Приветствую!
Программа стала работать.
В одно время скачал Candle_1.1.7 здесь- https://github.com/grbl/grbl/wiki/Using-Grbl
  и Grbl_v1.1f.20170801.hex здесь- https://github.com/grbl/grbl. 
В Arduino Uno R3 Atmega328P Atmega16U2 залил  прошивку Grbl_v1.1f.20170801.hex через  XLoader.
Открыл Candle_1.1.7 настроил необходимые мне параметры.
Проблемы  от  on 21 Jan  не стало. 

Artcl
 Каким постпроцессором пользуетесь?
 У меня такая же ошибка, также себя ведет программа. Я использую постпроцессор G-Code (mm) (*.tap) . 

 Перепробовал много постпроцессоров не нашел чтоб без ошибки отработала программа Candle.
 Каким постпроцессором пользоваться?
 Как сделать паузу для разгона шпинделя перед началом работу программы, не удобно включать шпиндель, ждать когда разгонится, потом включать программу на выполнение.

SadovovAlex, 2018-06-19:
ae1978 Написал на почту.
Denvi молодец, спасибо.

zdanat, 2018-07-17:
Здравствуйте, возможно обращаюсь немного не по теме, уж простите, подскажите пожалуйста, как решить проблему. Есть станок на Ардуино Уно+шилд v3 с драйверами 4988 . В начале работал без особых проблем, только кнопка home не отдавалась, в последнее время стал мой станок останавливаться сам собой, сначала перестают меняться значения в окошках правого верхнего угла, затем перестает двигаться отображения фрезы в окне визуализации, потом прекращает работать автопрокрутка и в заключение вообще станок перестает ездить. Где искать проблему, в кодах я не силен и поэтому вообще растерялся. спасибо.

caha200762, 2018-08-02:
Здравствуйте! 
1. Попробуйте стереть  Arduino , затем установить на нее заново   Grbl.
2. Сотрите с компьютера управляющую программу, программа которая управляет станком.
3. Установите управляющую программу заново на компьютер.
После этого проверьте работоспособность станка.

1. Какой у Вас станок, модель?
2. Какую версию использовали  Grbl ...  ?
3. Какую используете управляющую программу?

> Здравствуйте, возможно обращаюсь немного не по теме, уж простите, подскажите пожалуйста, как решить проблему. Есть станок на Ардуино Уно+шилд v3 с драйверами 4988 . В начале работал без особых [...]

zdanat, 2018-08-03:
Доброго времени! спасибо за отклик, уже пробовал перепрошивать, вначале
стояла грбл 0.9 и прога свеча версия 1.0. Прошил на грбл 1.1 и свечу взял
последней версии, все равно останавливается. Станок самодельный. Сегодня в
строке статус выскочила красная надпись тревога и станок завис, еще заметил
при нажатии кнопки домой станок никуда не едет а в консоли выскакивает
error 9. уже менял кабель и порт подключения - результат тот же. завтра
буду пробовать переставить виндовс, больше пока не знаю решений.

caha200762, 2018-08-19:
Доброго времени!
zdanat,  решились ли проблемы?
Может проблема в настройке Grbl?

Enabler88, 2021-06-09:
Здравствуйте, что за ошибка 15 может быть при управлении с нампада? Прошивка 1.1, версия 1.2.13b. Причем настроил кнопки в программе - все заработало, презапустил и все, ошибка 15 при попытках перемещения осей

closed 2025-09-14 by github-actions[bot]