Denvi/Candle #112: Error WinXP SP3 "The procedure entry point strnlen could not be located in dynamic link library msvcrt.dll"

prulitarit, 2017-01-23:
When i run Candle 1.1 on Windows XP SP3 with vcredist_x86 2010 installed i got errors "The procedure entry point strnlen could not be located in dynamic link library msvcrt.dll". 

Please help me to run latest Candle on Windows XP.

Denvi, 2017-01-23:
Install vcredist_x86 2013.

prulitarit, 2017-01-23:
already installed vcredist_x86 2013. And no result! Still shows error "The procedure entry point strnlen could not be located in dynamic link library msvcrt.dll". 

What version of Qt did you used in windows xp x86 ?

Denvi, 2017-01-23:
Qt 5.5.1

Denvi, 2017-01-23:
Было здесь #99. 
Смотрите в сторону драйверов/поддержки OpenGL 2.0. Windows XP c ANGLE не работает.
Для простых задач можно включить программную отрисовку, для чего нужно удалить файл "d3dcompiler_47.dll"

closed 2017-01-25 by Denvi