Denvi/Candle #113: Shifting map

leogala, 2017-01-29:
Hello,
I'm using this fantastic sender for an home made laser engraver and I'm really happy for the results I get.The only issue is that I have a probe not aligned with the laser, in particular it has a negative offset on y. 
Currently after the scanning procedure I simply move the y axes and send a G92Z0 command. Would be possible, once the map is buit to shift it respect to the work coordinates to align it with the laser ?

Denvi, 2017-01-29:
Hi.
Why not using G92Y0 also? Heightmap is consist of work coordinates, Z coords are relative to Z in X = 0 & Y = 0.

leogala, 2017-01-29:
Indeed, I use G92Y0 (was a typo, sorry). My request was to shift the map INSTEAD of moving the y axes. This would be useful because sometime I don't have the space to move back the head as I'm close to physical zero and I cannot move back...

leogala, 2017-01-29:
Discard the request. I solved by shifting the map start.

closed 2017-01-29 by leogala