Denvi/Candle #115: Нужна помощь

Kurmankozhoev, 2017-02-10:
Здравствуйте !
Мне нужен исходный код самой последней версии Candle 1.1.7   здесь скачал последнию версию исходного кода но оно отличается от версии Candle 1.1.7 например нет кнопки стоп ручного перемещения как здесь [image] вообщем не могли бы вы выложить исходный код последней версии программы для Linux ?

Kurmankozhoev, 2017-02-10:
У меня проблемы при компиляции подскажите в чем проблема [image] [image]

Denvi, 2017-02-12:
>например нет кнопки стоп 

Кнопка "стоп" есть только в ветке "grbl_1_1".

>проблемы при компиляции 

setPixelColor был заменен в одном из недавних коммитов, обновите рабочую копию исходного кода до последней версии.

Kurmankozhoev, 2017-02-13:
@Denvi  Огромное вам спасибо, все получилось )

closed 2017-03-04 by Kurmankozhoev