Denvi/Candle #116: Прошу смиренно помощи .

Lenivets71, 2017-02-10:
Подскажите пожалуйста. программа версия 1.1.7 прошивка 1.1 .Версия драйвера CH340 3.4.2014.8 от 8.8.2014 г и там и там Виндовс ХР сервис пак 3 . с право есть перемещение в ней ставлю Шаг Непрерывно , подачу любую . начинаю перемещать любую ось , то на стационарном компе (то есть управление с стационарного компьютера) происходит дёрганье станка .Если Шаг выбрать фиксированным (0.01, 0.1, 1, 5, 10, 100) то перемещение идёт гладко без дергатни. А на ноутбуке всё прекрасно, такого нет.
Забыл сказать выполнение G-кода идёт нормально .

Kurmankozhoev, 2017-02-10:
Обновите до версии 1.1.7 а прошивку до 1.1 и переустановите windows xp на windows 7 или 10 у нас была такая же проблема обновили прошивку до 1.1 и все нормально пошло.

Kurmankozhoev, 2017-02-10:
Если у вас есть делитель импульсов на драйверах , то ставьте не мение 64 тогда у вас всего это не будет( то есть дергания быть не должно ) двигателя будут работать гладко.

Lenivets71, 2017-02-10:
Прошивка и версия программы что Вы пишите у меня и так свежия программа Candle_1.1.7 прошивка Grbl 1.1e  Меня смущает тот факт что на старом ноутбуке всё работает цевильно, а на стационарном нет.

Kurmankozhoev, 2017-02-10:
Тогда я думаю на старом пк вам надо переустановить Windows на 7ку  возможно проблема в этом потому что программа написана на Qt creator там требования выше 7ки.

Lenivets71, 2017-02-10:
спс буду пробывать 7

Kurmankozhoev, 2017-02-10:
Совет лучше установить Linux если  у вас проблемы на винде с лицензей

Lenivets71, 2017-02-10:
линукс это понятно . всётаки грызёт этот вопрос. на ноуте и на стационарнике винда с одного диска , как так на ноуте всё без проблем а на стаци-ке такая трабла. непойму

Kurmankozhoev, 2017-02-10:
А вы установили microsoft visual c++  runtime library на станциоарке ?

Lenivets71, 2017-02-10:
что это такое и с чем его едят.?
как посмотреть установлен он или нет??

Lenivets71, 2017-02-10:
на ноуте установлены microsoft visual c++ (2005, 2008. 2010) на стационарном вечером гляну.

Kurmankozhoev, 2017-02-10:
Microsoft Visual C++ runtime – это один из тех пакетов, без которых на вашем компьютере не запустится большинство программ. Он содержит библиотеки, которые позволяют запуститься программам, в разработке которых была использована среда Microsoft Visual C++ и QT, известная так же как MSVC (Qt Creator). К таким приложениям относится большинство программ на Windows, потому наличие данного пакета необходимо....Ps забыл сказать есть еще Framework его тоже надо установить

Lenivets71, 2017-02-10:
Microsoft Visual C++ утановлен на стац. (2005,2008,2010,2012,2013,2015) а вот Framework только базовый 1.1. сейчас установлю 4.0

Lenivets71, 2017-02-18:
Вышел из положения установив 7-ку

closed 2017-03-18 by Denvi