Denvi/Candle #123: Visualization doesn't work - Win10 Intel(R) Q45/Q43 Express Chipset (MS Corp - WDDM 1.1)

rkreiple, 2017-02-20:
2 PC's, no visualization. Windows 10 64Bit

3rd same PC with NVidia GT610 works.

Apparently the Intel(R) Q45/Q43 Express Chipset (Microsoft Corporation - WDDM 1.1) driver won't handle it,  Driver is the latest.

[image]

rkreiple, 2017-02-28:
Just so you know, it was easier and faster to throw a $30 GT610 video board into the computer that I need it to work on than waiting for anyone to address this.

Opensource is nice, but this is bullshit.

Denvi, 2017-03-18:
Visualization problems was discussed many times. If your chipset supports OpenGL 2.0 you only need to install drivers from INTEL and use latest version of Candle (not grblControl as on screenshot).