Denvi/Candle #124: Программа закрывается с ошибкой, если выбрать растр

Rom327, 2017-02-25:
Здравствуйте! Нужна помощь! 
ОС Win7 x32 + все обновления. До недавнего времени пользовался версией 0.8.4. Все работает! Хотел перейти на версию 1.0.11(10) из-за добавления "растра" в визуализатор и возможности продолжения работы с произвольной строки.
1. При запуске 1.0.11(10) и загрузке программы, в визуализаторе наблюдается картинка в полоску, так же как и в версии 0.8.4:
[image]
 После нажатия "отправить" выскакивает окно с ошибкой. 
[image]
Но после "пропустить", программа начнет выполняться.
2. Если в настройках  визуализатора выставить "растр", картинка выглядит нормально 
[image]
**Но программа закрывается сразу после нажатия кнопки "отправить".**  Можно что-либо сделать?

Rom327, 2017-02-25:
Вот, что выдает windows:
[Ошибка.txt](https://github.com/Denvi/Candle/files/801382/default.txt)

Rom327, 2017-02-26:
Есть какие предположения?

Denvi, 2017-03-18:
Ничего не понял про то, что "выдает Windows".
Как именно программа закрывается?
Ошибка возникает с любыми растровыми файлами?

Rom327, 2017-03-18:
Спасибо, что откликнулись! Снял два видео, как закрывается программа. На разных картинках...
https://youtu.be/3opuLpvnWJ8
https://youtu.be/MdQitJu38HE

Denvi, 2017-03-18:
Боюсь без удаленного помощника Windows решить проблему будет сложно.
Можете выслать приглашение на почту denis_vic@mail.ru с указанием времени готовности к работе.

Rom327, 2017-03-18:
То есть Вы считаете, что проблема в компьютере, а не в программе? Попробую проверить эту версию на другом ПК. О результатах отпишусь.

Denvi, 2017-03-18:
На моем ПК таких проблем не наблюдается.

Rom327, 2017-03-18:
Спасибо, попробую. Может система подзабита...

Rom327, 2017-03-18:
И снова здравствуйте! Подключился к другой машине. Win7_64. Точно так же программа закрывается! Значит дело не в моем железе. Может какой библиотеки не хватает?

Denvi, 2017-03-18:
ОС тут как раз роль играет небольшую, куда важнее такие характеристики как:
- тип процессора/чипсета;
- тип видеокарты;
- объем ОЗУ.

Еще возможные причины:
- особенность отправляемых G-код файлов;
- настройки программы;
- настройки прошивки.

Rom327, 2017-03-18:
Я вас понимаю, но железо используется тоже, что и с версией 0.8.4. И еще, если в визуализаторе выбрать "вектор", то программа работает!

Denvi, 2017-03-18:
Версия 0.8.4 растры не рисовала, а код отрисовки он, кхм, несколько отличается от векторов.
У вас файлы загружаются, если программа настроена в растровый режим, загрузка происходит без ошибок только с векторным?
В растром режиме после загрузки файла программа не вылетает, если изменить текущую строку кода (поставить курсор примерно на середину файла, конец файла) в окне под визуализатором?
Если нажать кнопку "Сброс"?

Rom327, 2017-03-18:
Попытка редактирования программы  и программа закрывается!

Rom327, 2017-03-18:
Если редактировать в начале программы, то все работает, если в конце, программа вылетает!

Denvi, 2017-03-18:
Ясно, три вопроса - два ответа, и ни один не попал.
Вы одного из своих "пушистых" переслать можете?

Rom327, 2017-03-18:
А как тут слать? Или на почту...

Denvi, 2017-03-18:
Почтой будет проще всего.

Rom327, 2017-03-18:
Так какой адрес?

Rom327, 2017-03-18:
[Cat2.txt](https://github.com/Denvi/Candle/files/852603/Cat2.txt)

Rom327, 2017-03-18:
Заслал младшего кота. Прожжено несколько раз на 0.8.4.

Denvi, 2017-03-18:
Замечательно, ваш кот и вызывает эту ошибку. Буду разбираться. Спасибо.

Rom327, 2017-03-18:
Мой кот хороший!

Rom327, 2017-03-18:
Может  Вам имеет смысл пообщаться с автором программы, которая генерит УП?

Denvi, 2017-03-18:
Кот - хороший, а вот УП - нет. Удалите из вашего файла первую строку:
~~~
% 
~~~

Разбираюсь, почему она роняет программу...

Rom327, 2017-03-18:
Убрал %, все равно вылетает....

Denvi, 2017-03-18:
Редактирование заработало?

Rom327, 2017-03-18:
Убирал в блокноте, в начале и в конце программы.

Rom327, 2017-03-18:
В теле GRBL не дало отредактировать, вылет программы.

Denvi, 2017-03-18:
Все понял, завтра отпишусь по результатам.

Rom327, 2017-03-18:
Спасибо!

Denvi, 2017-03-18:
Замените исполняемый файл на тот, что в архиве: https://github.com/Denvi/Candle/releases/download/v1.0/Candle_iss124.zip
Все должно работать.

Rom327, 2017-03-18:
Заменил в версии 1.1.7. Без вылета! Спасибо! Завтра проверю на гравере!

Rom327, 2017-03-19:
Спасибо за программу! Проверил на гравере. Все работает! Единственное, что хотелось бы увидеть, так это подписи к осям:
[image]

Denvi, 2017-03-19:
Пожалуйста.

Подписи есть во всплывающих подсказках.

[image]

Rom327, 2017-03-19:
Не совсем удобно: надо мышкой наводить. Если бы надписи были, достаточно бросить беглый взгляд, да и  места достаточно чтобы разместить надписи.

Rom327, 2017-03-19:
И вот еще какая ерунда: при замене экзешника меняется надпись "подача" на "feed" и соответственно "переопределить" на "override".

Denvi, 2017-03-19:
Навели - запомнили, XYZ. Зачем каждый раз то смотреть на них? Или беглым взглядом не цепляется сразу три поля?
Места недостаточно в текущем интерфейсе, надписи - избыточны.

"Экзешник" для версии 1.0.11, там файлы переводов другие.

Rom327, 2017-03-19:
Понятно! На 1.0.17 и последующие версии планируется экзешник с исправлениями?

Denvi, 2017-03-19:
Чуть позже сделаю релиз для версий 1.0.12 и 1.1.8.

Rom327, 2017-03-19:
Спасибо! И,  с праздником!