Denvi/Candle #125: Error while loading shared libraries: libgobject-2.0.so.0

space55, 2017-03-03:


Denvi, 2017-03-18:
Candle doesn't tested on Linux x64, think, it should be recompiled for your system from sources.

Nandox7, 2017-03-20:
Summarizing  as Dennis mentions the binary is for i386 not for x64. That said I managed to make it run on my x64 Linux instalation. 

Force to install the i386 version of the package. 
apt-get install  libgobject-2.0.:i386

In my case I had to install a few more things but is due to the distribution I use elementary OS, that doesn't use Gtk themes and the default Gtk theme for QT couldn't be found,  breaking a bit the UI.

Stas2402, 2020-11-08:
where i can find sources?

Hi-taker, 2022-06-18:
Сам заебался. 
Автор, сделай нормальную инструкцию для Linux отведения!

where i can find sources?---->>>
sudo dpkg --add-architecture i386
sudo apt update

son1csun, 2024-06-13:
@Nandox7  Unfortunately, these packages aren't available for Ubuntu 24.04.  

user27@user27-HP-250-G8-Notebook-PC:~/Candle_1.1.7/Candle$ sudo apt install libgobject-2.0.:i386
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libgobject-2.0.:i386
E: Couldn't find any package by glob 'libgobject-2.0.'

I'm trying to install CAndle_1.1.7 on a 64 bit machine and am getting this error.  

./Candle: error while loading shared libraries: libgobject-2.0.so.0: cannot open shared object file: No such file or directory

I've installed all of the gobject libraries, but Candle still won't start.  

@Hi-taker solution also doesn't work.

mar0x, 2024-06-24:
@Nandox7 Try this build [Candle-1.1.9-Linux-x86_64.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-Linux-x86_64.AppImage)

closed 2025-09-03 by github-actions[bot]