Denvi/Candle #129: Missing File?

pmbrown141, 2017-03-16:
Version 1.1.7, on startup, error MSVCP120.dll not found. Upon clicking "OK" the app starts, shows connected, but jog buttons inop.

Kurmankozhoev, 2017-03-17:
@pmbrown141 

> Version 1.1.7, on startup, error MSVCP120.dll not found. Upon clicking "OK" the app starts, shows connected, but jog buttons inop.

> 
In case you receive this error message regarding a missing DLL file, you have to install the Microsoft Visual Studio 2013 Runtime Libraries. Download them here:[https://www.microsoft.com/ru-RU/download/details.aspx?id=40784](url)

pmbrown141, 2017-03-18:
The link takes me to a site in a language I'm not familiar with. I'll search for the same in English.

electrokean, 2017-03-18:
@pmbrown141 That link was for the Russian MS download page.
The US English link would be https://www.microsoft.com/en-US/download/details.aspx?id=40784

pmbrown141, 2017-03-25:
OK, so now I have Studio Runtime Libraries installed as advised, and now I get an error "The procedure entry point strnlen could not be located in the dynamic link library msvcrt.dll", several times when starting the program.

electrokean, 2017-03-25:
Are you running on Win XP?
It is a strange error message, but a Google search indicates that XP is no longer supported by Qt & ANGLE which are required for Candle
For example, see https://github.com/Denvi/Candle/issues/112 and http://stackoverflow.com/questions/36105432/qt-creator-on-windows-xp-the-procedure-entry-point-strlen-could-not-be-located

closed 2025-09-03 by github-actions[bot]