Denvi/Candle #131: Jog not working on version 1.1.7

delimaras, 2017-03-20:
If i click any direction of jog not response but homing cycle is work. In old versions 1.0.11 or grbControl all is work perfect.

iwoj, 2017-03-20:
I can confirm this. Manual spindle control works fine but x, y, z jog doesn't.

Denvi, 2017-03-21:
Hi, please provide:
1) firmware version;
2) firmware settings;
3) Candle settings.

iwoj, 2017-03-21:
Firmware: Grbl 0.9j
Settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Candle settings: default (COM3)

Denvi, 2017-03-21:
Candle 1.1.x works only with GRBL 1.1.
Use Candle 1.0.x or update your firmware.

minyor, 2017-12-17:
I've updated to GRBL 1.1 but Jog still do not working.
Also why do you guys stripped off step presets (0.01, 0.1, 1, 5, 10, 100) from new versions?
This was very handy((

andired, 2018-01-22:
Hello Denvi,

It seems I have the same issue with Candle 1.1.7 as well. Jog doesn't work at all.
Tested with other software and the hardware works ok.
Below are my settings

[CTRL+X] < Grbl 1.1f ['$' for help]
$ < [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok
$$ < $0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

Kim9999, 2018-02-10:
[settings Candel 1.1.07 original.txt](https://github.com/Denvi/Candle/files/1712899/settings.Candel.1.1.07.original.txt)
Yup me too cannot seem to get anything to move in jog.  3 days of reading and entering code to file,  the thing says its running but nothing moves! The $J= is not in ther code at all. It would be nice if someone had a sampke of the code to get this running,  I have read and re read the jogging section but have not got the code correct yet.
[settings Candel 1.1.07 original.txt]