Denvi/Candle #133: Candle 4 Axis

Kurmankozhoev, 2017-03-24:
Здравствуйте !
Хотел спросить у вас, а Candle поддерживает 4 axis ? просто мы перешли на mega 2560, так как mega 328 у него мало памяти и 4 axis добавить не получится, по этому не могли бы вы добавить 4 axis в Candle ?   Есть еще [вот](http://savepic.ru/13339439.png) эта программа поддерживает 4 axis, но программа не для всех ОС а мне надо и на Linux и на Windows, программа ваша очень понравилось и по этому так сильно хотим что бы машина работала именно на нем, так как Candle боле надежная а другие программы ломаются причем быстро, и даже ошибок не выдают.

Denvi, 2017-03-27:
Здравствуйте, не поддерживает и поддержка не планируется.
Спасибо.

Kurmankozhoev, 2017-04-02:
@Denvi  а посоветовать можете как добавить 4 Axis ?

selenur, 2017-04-02:
Исходники данной программы доступны, вам нужно найти программиста, который согласился-бы доработать программу под ваши требования.

closed 2017-04-17 by Kurmankozhoev