Denvi/Candle #134: Open source 1.1.7

tuananhproed, 2017-03-28:
Hello Denvi 
I have upgraded grbl 1.1f and use open source of candle 1.1, but unfortunately it's not a new version and there are many bugs, i can only download bundled software, i use source code and Want to add a cool off switch, can you share the source code of Candle 1.1.7?
[image]

Denvi, 2017-03-28:
Hi, latest code for GRBL 1.1 is in "grbl_1_1" branch. Check it out please.

tuananhproed, 2017-03-28:
Hi, i got the grbl 1.1f version upgraded to my cnc, the candy 0.9 source code i rebuilt has not worked, the current candle 1.1 source code is not working well in jog mode, i It has been changed in Candle 1.1.7, please let me know how you did, I just want to make a few changes to interface 1.1.7 and add coolant toggle. I'm not a professional programmer. Sorry my english is not good
[image]
[image]
[image]

tuananhproed, 2017-03-28:
I am very happy to find, thank you

closed 2017-03-28 by tuananhproed