Denvi/Candle #135: Point & Click to move to position.

Nandox7, 2017-03-28:
Hi,

It would be good to have an option to click in a given point in the visualization and have the cnc move to that point.
Main use case I have for this Is when re-using plates to see if what I'm about to cut will fit and adjust the home position accordingly.

Thanks!

Konesuke, 2017-03-31:
Yes it is a good idea! But it is necessary also a line to define the machine max area!

stikdragon, 2017-08-23:
This would be really useful, although to prevent accidental clicking make it so you need to either be in a one-shot mode, or holding a modifier key (ctrl, perhaps)

Ramzezka, 2017-12-24:
+1.
I need that thing for make pcb.
Now i need to set zero level upper and start work, wait untill drill will be on needed hole and adjust zero X Y.

cinguecento, 2019-08-21:
How to reset the point

closed 2025-09-03 by github-actions[bot]