Denvi/Candle #136: Help button Cool Water

tuananhproed, 2017-04-01:
Hello Delvi
I tried adding a button on the control panel to turn off the cooling water, which was very convenient during operation. But I can only make it display when hit, the command sent from the command prompt does not show, I want it to behave like Spind control button? Help me do that

In addition, how can I display the status of the button when off = black, open = red?

This is the command I use

`void frmMain::on_cmdCoolwater_toggled(bool checked)
{
     sendCommand(checked ? QString("M8") : "M9", -1, m_settings->showUICommands());
}`

Denvi, 2017-04-01:
In "frmmain.cpp" from line 1062:

~~~
                // Process accessories state
                static QRegExp as("A:([^,^>^|]+)");
                if (as.indexIn(data) != -1) {
                    QString state = as.cap(1);
                    
                    // Spindle
                    m_spindleCW = state.contains("S");
                    if (state.contains("S") || state.contains("C")) {
                        m_timerToolAnimation.start(25, this);
                        ui->cmdSpindle->setChecked(true);
                    } else {
                        m_timerToolAnimation.stop();
                        ui->cmdSpindle->setChecked(false);
                    }

                    // Coolant
                    ui->cmdCoolwater->setChecked(state.contains("F"));    // "M" - for mist coolant
                    
                    if (!pinState.isEmpty()) pinState.append(" / ");
                    pinState.append(QString(tr("AS: %1")).arg(as.cap(1)));
                } else {
                    m_timerToolAnimation.stop();
                    ui->cmdSpindle->setChecked(false);
                    ui->cmdCoolwater->setChecked(false);
                }
~~~

> void frmMain::on_cmdCoolwater_toggled(...

replace with realtime command:

~~~
void frmMain::on_cmdCoolwater_clicked(...
{
    m_serialPort.write(QByteArray(1, char(0xa0)));
}
~~~

Icons can be setted on each state (checked, unchecked) in form designer.

tuananhproed, 2017-04-01:
I followed the instructions, everything was fine, but when I pressed the Coolant button, the command line did not display, I tried adding the old command line, it was right for my purposes, but I still do not understand how to do. :) , I'm not a programmer, everything I copied from the Spind command

`void frmMain::on_cmdCoolwater_clicked(bool checked)

{

    m_serialPort.write(QByteArray(1, char(0xa0)));
    sendCommand(checked ? QString("M8") : "M9", -1, m_settings->showUICommands());

}`

Denvi, 2017-04-01:
Realtime command "0xa0" will not display "m8" or "m9" in console, but will work in "hold" "idle" & "run" state properly.
You can use "m8" & "m9" instead. But in this case remove
~~~
m_serialPort.write...
~~~ 
line.

Check: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

tuananhproed, 2017-04-01:
Thank you Denvi for the guidance, have a nice day  :)

closed 2017-04-01 by tuananhproed