Denvi/Candle #137: P (dwell command in .nc files) msec

Simplystupid, 2017-04-05:
I have P5000 in my gcode. From Techne CAD CAM. Author assures me this is msec not seconds, like FANCU or whatever industrial standard.
If I change to P5, sequence dwells, no action, 5 seconds.
So Candle 1.1.17 latest version, treats digits following P as seconds. 5000 seconds is like, forever.
Bit of a pain to keep having to open .nc file in text editor and change the values.

gerritv, 2017-04-05:
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4, it is seconds

Simplystupid, 2017-04-06:
Yes, thanks, that is what I thought but the guy on Techne CAD CAM has it down as msecs. I know there are some other U codes with msecs Uxxxx. I've asked if he can change his default P5000 with variable speed spindle selected.

closed 2025-09-03 by github-actions[bot]