Denvi/Candle #140: grbl 0.5.2 w7 bit32

2476, 2017-04-16:
Hello, I do not understand why when I click the "return x, y, z" icon, the z-axis does not return to the starting point. I run the "zero  z-axis" operation successfully. When carrying out the return, the x and y axes go to the starting point, the z axis does not move. The procedure "home" is carried out successfully, all the functions of the program are active, the operation of the device is correct. Only this icon does not give me rest.

2476, 2017-04-16:
Can you communicate in Russian?

Denvi, 2017-04-18:
В настройках программы имеется флаг "Перемещение по " Восстановить СК" в ..." и рядом выпадающий список с пунктами "плоскости", " пространстве". При выборе второго пункта перемещение будет производиться по всем трем осям.

2476, 2017-04-18:
> Кому: "Denvi/Candle" <Candle@noreply.github.com>
> Копия: "2476" <komi2476@yandex.ru>; "Author" <author@noreply.github.com>
> Отправлено: 18.04.2017 5:50:36

см.вложение Что то не могу найти данные пункты....
------ Исходное сообщение ------

2476, 2017-04-18:
[image]

Denvi, 2017-04-18:
Верно говорите. В версии 0.5.2 нет такой настройки.
Обновиться до 1.0.x есть возможность?

2476, 2017-04-18:
Да.