Denvi/Candle #143: Problem with large file size

tuananhproed, 2017-05-05:
Hello Denvi
I have a problem with large files, when starting to load into the program file an error message occurs, can you help me know the cause or how to fix it. Thank you.
[image]

Denvi, 2017-05-08:
Hi, how many lines your file has?
"Candle" doesn't optimized for low memory usage, it's consumes about 2,5 Gb with 5 million lines g-code file.

tuananhproed, 2017-05-09:
Thank you Denvi for the positive, initially my nc file size was about 75Mb, I thought it did not fit and split the program into 5 steps, now I have 5 files, each file is small More than 20Mb and it works well.
However, there is one inadequacy that the Candle program does not automatically run the files one by one, I think you go up there menu to run programs in turn for the next update, Thank you.

closed 2017-05-10 by tuananhproed