Denvi/Candle #145: Можно ли автоматизировать обнуление?

Rom327, 2017-05-14:
Здравствуйте! Можно ли сделать так, чтобы после выполнения команды домой ($H), автоматически обнулялись рабочие координаты? Спасибо!

Denvi, 2017-05-14:
Можно, но делать я этого конечно же не буду.

P.S. Создайте пользовательскую команду:
~~~
$H
G92 X0 Y0 Z0
~~~

Rom327, 2017-05-14:
Жаль! Может в следующих версиях добавите такую возможность? Или возможность редактировать кнопки. А в целом, спасибо за программу!

Rom327, 2017-05-15:
П. С. Создайте пользовательскую команду:
$H
G92 X0 Y0 Z0
Не работает! При этой команде рабочие координату становятся равными машинным. А мне бы хотелось, чтобы рабочие координаты были равны 0.

Denvi, 2017-05-15:
О как, опишите подробнее всю процедуру по шагам, как вы это себе представляете. Какие координаты и когда должны быть. Попробую разобраться.

Rom327, 2017-05-15:
Спасибо, что ответили! Что бы мне хотелось: при нажатии кнопки "домой" выполняется команда $H и далее сами обнулялись рабочие координаты. Сейчас же приходится нажимать 3 кнопки: "домой", "Х0 У0", "Z0". Оно как бы не лениво, но очень часто забываю обнулить  и запускаю программу. Соответственно инструмент летит по G0 в машинный ноль, а потом врезается в опору. [https://youtu.be/9BJJar9AH3E](url)

Denvi, 2017-05-15:
Хорошо, если после команды $H дать команду G92X0Y0Z0 разве рабочие координаты не обнулятся?
Другое дело, что они не запишутся в память программы и кнопка "Восстановить СК" не будет корректно работать. Но это исправимо.

Rom327, 2017-05-15:
Спасибо, за ответ! Я разобрался в чем была причина: я написал команды на свободную кнопку "1" $H G92 X0 Y0 Z0. При этом пропустил ";" после $H. Сейчас исправил и команда выглядит так: $H; G92 X0 Y0 Z0. Все заработало! Но все же хотелось бы иметь возможность редактировать команды всех кнопок, как это сделано для кнопок "Z-щупа" и "безопасного положения". Может добавите в будущие версии такую возможность?
Спасибо!

Denvi, 2017-05-15:
Да, надо добавить.

Rom327, 2017-05-15:
Спасибо!

grblkit, 2017-10-18:
> Другое дело, что они не запишутся в память программы и кнопка "Восстановить СК"
а как это исправить?
командой G9* X0 Y0 Z0 ?
здесь нет кстати G92 https://ru.wikipedia.org/wiki/G-code

selenur, 2017-10-19:
G-коды лучше тут смотреть: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92

Sergeyvp2018, 2018-03-11:
Здравствуйте!
Обновил прошивку китайского CNC 2418 до grbl_v1.1f.20170801
Candle 1.1.7
Не работает кнопка домой, выдаёт ошибку   $H < error:5
И не работает кнопка восстановления СК, пишет всё ок, но ничего не происходит.
Как с этим бороться?

closed 2025-09-03 by github-actions[bot]