Denvi/Candle #146: Развитие программы в дальнейшем

Sergeulyanko, 2017-05-15:
Здравствуйте, Денис
Планируется ли развитие программы? В данный момент собираю 5 - осевой фрезер. Есть ли у вас интерес в доработке программы под 5-осевую обработку? Рассматривает варианты взаимовыгодного сотрудничества? Почта для связи sergeulyanko(собачка)gmail.com

Denvi, 2017-05-15:
Здравствуйте, спасибо за внимание к программе.
Боюсь, не будет развития толком, ибо высокая занятость на основной работе.

Rom327, 2017-05-16:
Денис! Не бросайте программу! Программа супер!!! Спасибо!

yuranikspb, 2017-08-09:
Нет! Не вздумайте бросать программу!

Peter6891, 2017-09-09:
А если совсем уж в будущее, - то очень не помешало бы под токарный станок по металлу заточить... Ибо для токарных с ЧПУ софт практически отсутствует...

closed 2025-09-03 by github-actions[bot]