Denvi/Candle #147: Linux compiling steps/instructions?

Nandox7, 2017-05-15:
I was trying to be able to compile the source to test some changes, but I the current master branch or even the grbl_1_1 don't produce the same version as the one shared in the releases, for example the continuously jog movement is not present and the probe option does not work.

Am I missing something here?

PS: I see these instructions are missing in the wiki, I'll be glad to help and add them if there is any interest.

MockbaTheBorg, 2017-05-16:
Hi There. I would love to build form source on Windows.
Do you have any instructions or step-by-step?

Thanks!

Nandox7, 2017-05-16:
So far what I've gathered is that you need to install QTCreator from QT.
https://www.qt.io/download-open-source/

Latest version do already bring the SerialPort library included so no need to install it.
For windows I'm not sure as I haven't tried, but for linux there is some extra steps needed due to missing libraries.

After having QTCreator installed open with it the candle.pro file inside the src/ folder.
If all libraries are in place it should compile/run directly.

bazhenov, 2017-05-19:
I was able to build application on macOS using two simple steps:

1. running `qmake` inside `src` folder;
2. then running `make`.

qt 5.8.0

guillaumepicquet, 2017-12-04:
I found a way to build it for Ubuntu 16.04 LTS from scratch.
I cloned the project.
`git clone https://github.com/Denvi/Candle.git`
Install QT5
`sudo apt-get install qt5-default qttools5-dev-tools libqt5serialport5-dev`
`cd Candle`
`qmake -project`
`qmake`
Now there is a _Makefile_
`make`
`./Candle`

All colors default to black I changed the settings to fix.

guillaumepicquet, 2017-12-08:
Git master branch is not on grbl_1_1
So to build last version:
- Install QT5
`sudo apt-get install qt5-default qttools5-dev-tools libqt5serialport5-dev`
- Clone Candle and switch to GRBL 1.1 branch
`git clone https://github.com/Denvi/Candle.git`
`cd Candle`
`git checkout grbl_1_1`
- Run Qmake
`cd src`
`qmake`
- Build with the generated _Makefile_
`make`
`./Candle`

novel-yet-trivial, 2018-08-06:
@guillaumepicquet Thanks so much but you left a few things off of your install step. I had to use

    sudo apt-get install qt5-default qttools5-dev-tools libqt5serialport5-dev g++ libqt5opengl5-dev

guillaumepicquet, 2018-08-15:
@novel-yet-trivial Thanks for for the info, answer updated.

evgen48, 2022-04-25:
> All colors default to black so I changed the settings to fix.

thank you so much, after compiling qt creator my self, and fixing lots of qt6 related issues in candle source code, i've got those black gl widget, i thought something is failing, however there  just colors were defaulted to black, switching background to white fixed all issues.

closed 2025-09-02 by github-actions[bot]

limann77, 2025-09-06:
@evgen48 Do you have a working Qt6 fork?