Denvi/Candle #148: Add option to directly save heightmap.

Nandox7, 2017-05-15:
Hi,

I've been using the height map function and it works great but the workflow is a bit odd.
Most of the times I accidentally reset the current map as there is no clear option to save it.

Ideally I'd see it as.
1. Press create to go into edit mode
2. Within edit mode to be an option "Save" to save the current map to a file.

At the moment the only option I find to save it is by pressing create again and it asks to save it, but when it does that I need to open the saved file as it resets the current heightmap.

Hope this makes sense.

Denvi, 2017-05-15:
> Within edit mode to be an option "Save" to save the current map to a file.

This works exactly now. In edit mode choose "Save" or "Save as..." from "File"-menu.

It's bad practice to use modes, I know.

Nandox7, 2017-05-15:
Oh ok, I'll try that. I was just using the app menu for the initial settings. 
To be honest mine shows localized in Portuguese and the translation is a bit... bad.
So it makes it a bit more tricky to deal with. :)

closed 2025-09-02 by github-actions[bot]