Denvi/Candle #150: Problems with auto level

pedro06, 2017-05-25:
Hello, I have a query I have the soft in millimeter, I use a wick conic of 0.1 mm. When I do the altitude measurement gives me values in + and - from 0.001 to 0.018. Then I made the click to use height map and the modification is made of 0.100 which is the depth I gave you and I do not get a smooth job, it has different heights, my doubt is if you take the height measurement it is done in mm Or inches. Another question in State Coordinates of work and Coordinates of machine this marking me what? Mm, inches, would you have a manual to see the different options you have? . I appreciate the tremendous work you do with this soft. Greeting.

Denvi, 2017-05-26:
Hi.
Heightmap values in table - mm.
Status units - as you set in your grbl firmware and in the "Settings > CNC info > Units" setting.

> When I do the altitude measurement gives me values in + and - from 0.001 to 0.018. Then I made the click to use height map and the modification is made of 0.100

Need example of g-code program before and after applying heightmap with described problem.

pedro06, 2017-05-26:
([image])
[image]

I send you the original file, the corrected file and the map file:
[Pistas.txt](https://github.com/Denvi/Candle/files/1030760/Pistas.txt)
[Pistas_corregido.txt](https://github.com/Denvi/Candle/files/1030761/Pistas_corregido.txt)
[mapa1.txt](https://github.com/Denvi/Candle/files/1030762/mapa1.txt)

Denvi, 2017-05-28:
I can't download any files.

pedro06, 2017-05-29:
What a strange Denvi, recently probe from the pc and the cell and I was able to download the files, the only ones that are with extension txt and the originals like ngc. Could it be that you can not download them?

pedro06, 2017-05-29:
Denvi solved it, had badly configured GRBL in inches, passed it to mm and funsiona of wonders. In case someone else happens to you I leave you $ 13 = 0 mm $ 13 = 1 inch

closed 2025-09-02 by github-actions[bot]