Denvi/Candle #151: не открывает УП больше 10 Мб

AlexxxCNC, 2017-05-26:
При загрузке программы размером 30 Мб, candle зависает и показывает пустое окно визуализации и не показывает g-код (1 скриншот). Когда нажимаю "отправить", вылезает ошибка (2 скриншот). 
В чем проблема? подскажите, пожалуйста\
Версия 1.1.6. В версии 1.1.7 то же самое.
[image]
[image]

selenur, 2017-05-26:
Попробуй в G-коде убрать строки начинающиеся с символа двоеточие, так-же убрать нумерацию кадров.
Вот такой текст G-кода отображается нормально:

```
g0 f500 x0 y0 z0
x50
y50
z50
x0
y0
z0
```

AlexxxCNC, 2017-05-26:
сделал в настройках частоту кадров 30 и запустилось нормально.

Slukka, 2017-06-09:
Тоже такая же ерунда. Подскажите, а каким лучше форматом сохранять траектории в ArtCAM-е, а то их там такое множество.

AlexxxCNC, 2017-06-10:
1. Чтобы открывались большие уп, нужно в настройках визуализации поставить галочку "упростить геометрию". Это проблема только из за слабого компа, на хорошем производительном компе все открывается на ура. 
2. В Арткаме я сохраняю траекторию для Candle  обычно в формате mach2 cnc.

Slukka, 2017-06-10:
Да вроде комп не очень слабый. i5-4570, gtx750, 4gb ram, SSD 120Gb
Но галочка помогла, спасибо.

Peter6891, 2017-09-09:
я в Арткаме сохраняю в формате .tap

Peter6891, 2017-09-09:
G-Code (mm) .tap

closed 2025-09-02 by github-actions[bot]