Denvi/Candle #153: Ошибка при выполнении кода

AlaAndr, 2017-06-18:
Здравствуйте, помогите разобраться в проблеме. Во время выполнения программы появляются сообщения об ошибке.  Выбираю "Пропустить", работает дальше, затем опять ошибка.
[image]
Во время работы местами мерцает индикатор "Работа", очень быстро, еле успела прочитать, что появляется "Готов". 
 Результат выглядит так:
[image]
Почему это происходит?

Denvi, 2017-06-19:
Приветствую. Судя по всему, здесь имеются проблемы с передачей данных. Попробуйте заменить USB-кабель, подключить кабель к другому порту, увеличить период опроса состояния с 40 до 250 мс и выше.

AlaAndr, 2017-06-19:
Спасибо. Смена кабеля на то, что дома нашлось, помогла не очень. Попробую купить еще один, подороже.

closed 2025-09-02 by github-actions[bot]