Denvi/Candle #154: Try to build

frobgm, 2017-07-18:
Dear Denvi

I try to build your program , the source is oldiest than the last zipped version...

I download latest Qt and build, glwVisualiser window is black, I don't know why?
I like to litle modification the screen, but first problem the default build is wrong.
Please help me.
Thanks
Robert

Denvi, 2017-07-18:
Hello. Version 1.1.x is in "grbl_1_1" branch (master is for previous grbl). Check it out please.

> glwVisualiser window is black

Can you post screenshot?
Try to reset settings to defaults (to set default colors for visualizer).

frobgm, 2017-07-19:
Hello
[image]
about show the source version is 1.0.12
the latest zipped executable version is 1.1.7

Yes, you're right, settings default screen is black.

frobgm, 2017-07-26:
Dear Denvi
I try to modificate the screen, but I wrote to you , the master source files is oldiest, possibly you download latest source, or you modificate same the screen what I attached?
[image]

many many thanks!

Denvi, 2017-07-26:
Here is the latest sources: https://github.com/Denvi/Candle/tree/grbl_1_1

frobgm, 2017-07-26:
Thanks!
I seen this is 1.1.8 version, what's the difference compared to the 1.1.7 version?