Denvi/Candle #155: Система координат

halny123, 2017-07-19:
Здравствуйте !
Подскажите я немного запутался в координатах.
1. Правильно ли я понял, что начало системы координат у программы это левый нижний угол и направление осей вверх-право? (у меня станок с лазером оси Z нет)
по крайней мере на экране это так отображается. Настроил GRBL так, что при подаче команды домой ($H) станок движется в левый нижний угол, но при этом координаты машинного и рабочего нуля становятся отрицательными и соответствуют рабочему полю заданному в GBRL для софт лимита.
Правильно ли это ? Думал что при приходе в дом машинные координаты должны устанавливаться в 0.
2. Как сделать что бы в доме машинные и рабочие координаты становились 0 ?
3. Возможно ли изменять в программе систему координат (и направление осей) ?
4. Есть 4 программируемые кнопки, можно ли их активировать по быстрым клавишам ? если да то какими, а то не очень удобно переключаться на ноутбуке с маленьким экраном. 
5. Есть ли описание программы ?

Спасибо, из 3 испробованных программ эта понравилась больше всех, ей и пользуюсь но вот остаются вопросы.

Denvi, 2017-07-20:
Приветствую.
1) Верно понимаете. Направление вверх-вправо. В "доме" машинные координаты должны быть нулевыми (но для корректной работы достаточно правильного направления осей, т.к. работа осуществляется в рабочих координатах).
2) Читать инструкцию к прошивке GRBL.
3) Направление нет, начало рабочих координат можно (кнопки "Обнулить XY", "Обнулить Z", команда "G92").
4) Не реализовано.
5) Неполное в [wiki](https://github.com/Denvi/Candle/wiki).

halny123, 2017-07-27:
Спасибо за ответ.
Специалист с меня небольшой, поэтому есть много вопросов которые могут показаться смешными.
Т.к. станок собирал сам для лазера (2Вт) возникают вопросы на которые пытаюсь найти ответы.
После долгих поисков постпроцессора ArtСam GRBL Laser для работы с лазером по кодам М3 и М4 написал его сам. Если кому нужно можно взять [тут](https://drive.google.com/drive/folders/0B1XrAS43VHIiWWFRYWwtZHBnZ2M?usp=sharing).
Опробован с ArtСam12 - GRBL1.1f - Candle 1.1.7

А можно попросить добавить возможность активировать программируемые кнопки быстрыми клавишами, хотя бы в режиме ручного управления ?
Навесил на кнопки функцию включения/выключения лазера на минимальную мощность для позиционирования. И когда позиционируется лазер на заготовке приходится долго листать по боковой панели с кнопок на перемещения, на маленьком ноутбуке не удобно.

Спасибо.

closed 2025-09-02 by github-actions[bot]