Denvi/Candle #157: Problems with g-code G91?

domstar, 2017-08-11:
Dear Denvi,

is it possible that the g-code command G91 is not implemented at the moment or is there any error within it?
Thank you in advance! (especially fo this great progam :))

Best Regards
domstar

Denvi, 2017-08-24:
Idk about errors. Could you explain?

guillaumepicquet, 2017-12-11:
Hello Denvi,
> Could you explain?
I've just build Candle for Ubuntu, works really great. Thank you for the software !
With the following GCode tool path is drawn:
`G91`
`G00 X10 Y10`
But I found a workaround by inserting a absolute move before G91:
`G00 X0 Y0 Z0`
`G91`
`G00 X10 Y10`
Here the tool path is drawn.
Maybe there is  a lack of initialization somewhere.

Regards
Guillaume.

Denvi, 2017-12-28:
Hi.
Tool path should have full determined first point for now to right draw. For example, as you use before:

~~~
G00 X0 Y0 Z0
~~~

closed 2017-12-28 by Denvi