Denvi/Candle #158: 64-bit OS?

magpern, 2017-08-21:
I would like to run this on Ubuntu Desktop 64-bit (17.04). Would it be possible to compile this to 64-bit?
Sure, I could probably install multi-architecture on ubuntu and run the 32-bit on a 64-bit os, but it was not without errors on the first attempt.
I have no clue on how to set up a compile environment for this myself, since I am more of a .NET developer.

artchizhov, 2018-01-03:
I use Ubuntu 16.04.3 64bit, i solve problem by installuing this packages, try may be help.

sudo apt-get install libqt5libqgtk2:i386
sudo apt-get install libcanberra-gtk-module:i386
sudo apt-get install libatk-adaptor:i386 libgail-common:i386
sudo apt-get install overlay-scrollbar-gtk2:i386
sudo apt-get install gtk2-engines-murrine:i386
sudo apt-get install gtk2-engines-pixbuf:i386

Nelstaar, 2018-01-04:
Hi, I did build it for Ubuntu 64 Bits.
[check issue #147 ](https://github.com/Denvi/Candle/issues/147)

closed 2025-09-02 by github-actions[bot]