Denvi/Candle #160: If heightmap probing fails you are forced to reset the origin

stikdragon, 2017-08-23:
When it enters alarm state while in heightmap mode the only button that remains enabled is the Reset button.  Once you press it you can get back to the rest of the application, but now the origin has gone.

Denvi, 2017-08-24:
Buttons "Zero XY" and "Zero Z" stores your origin.
After reset you can restore by "Restore XYZ" button.

mjml, 2020-04-07:
Those buttons reset your working origin. The machine coordinates (for example under GRBL) will plunge to large negative values of Z as the stepper motor grinds away trying to plunge the bit into the copper. Once the grinding starts, I cannot seem to issue console commands to stop it. This can happen for example if your probe leads become loose from the Dupont terminals on your GRBL board.

At the end of the failed probing sequence, your bit is sitting on the pcb and the stepper has just finished grinding away. GRBL machine coordinates are Z < -15.0. Although you can reset the working coordinates, the machine coordinates are stuck there.

closed 2025-09-02 by github-actions[bot]