Denvi/Candle #161: В окне 3D просмотра отображаются не все движения станка

Elektron2016, 2017-08-24:
Видимо последние несколько движений станка обрезаются. Например в такой программе:
G21
G90
G94
F50.00
M03
G4 P1
G01 Z-2.5000
G01 X-13.5
G01 Y-41.5
G01 X0
G01 Y0
G00 Z5.0000
M05
Последние три движения не отображаются.

Denvi, 2017-08-24:
У меня отображается

[image]

Elektron2016, 2017-08-24:
А должно быть так [http://rgho.st/6JnL2L82q](url)

Denvi, 2017-08-24:
Не работает ссылка

Elektron2016, 2017-08-24:
Уже исправил.

Denvi, 2017-08-24:
Разбирайтесь, можно drag&drop'ом вставить картинку.

Elektron2016, 2017-08-24:
[image]

Denvi, 2017-08-24:
Теперь вижу.
На самом деле обрезаются начальные движения, вплоть до первой полностью определенной точки, а это:

~~~
G01 Y-41.5
~~~

Elektron2016, 2017-08-24:
Как тогда должна выглядеть программа чобы всё отображалось корректно?

Denvi, 2017-08-24:
Добавьте в начало

~~~
G4 P1
G0 X0 Y0 Z0
...
~~~

Elektron2016, 2017-08-24:
Спасибо, не знал таких тонкостей.

closed 2017-08-24 by Elektron2016